HelTecAutomation/CubeCell-Arduino #1: Failure in Arduino IDE

raystream, 2019-10-12:
i have install the cubecell via the board manager.
when i compile i get this error:

exec: "C:\\Program": file does not exist

Heltec-Aaron-Lee, 2019-10-13:
Did you choose a correct board? And did you try with our provided examples?

raystream, 2019-10-13:
> [
> {
> "@type": "EmailMessage",
> "@type": "ViewAction",
> "target": "https://github.com/HelTecAutomation/ASR650x-Arduino/issues/1?email_source=notifications\u0026email_token=ABR5JNAFYAYXSVR4ESXVEO3QOMU4JA5CNFSM4JADVTUKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEBCXFPA#issuecomment-541422268",
> "url": "https://github.com/HelTecAutomation/ASR650x-Arduino/issues/1?email_source=notifications\u0026email_token=ABR5JNAFYAYXSVR4ESXVEO3QOMU4JA5CNFSM4JADVTUKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEBCXFPA#issuecomment-541422268",
> "name": "View Issue"
> },
> "description": "View this Issue on GitHub",
> "publisher": {
> "@type": "Organization",
> "name": "GitHub",
> "url": "https://github.com"
> }
> }
> ]

HelloI habe choosen tue capsule as board and tried the lorawan example.

raystream, 2019-10-14:
i have found out the problem.

i have to install the arduino ide. Using the IDE in portable mode (creating a directory "portable" in the programm directory) is not functioning.

Heltec-Aaron-Lee, 2019-10-15:
OK, Can we close this issue now?

raystream, 2019-10-15:
HelloIssue can be closedRegards bjoern ---------------------------Björn A. BlöhdornWiesengrund 2224232 Schönkirchen0179-209824

[
{

Heltec-Aaron-Lee, 2019-10-16:
where is the message from? I don't think it's an Arduino message.