HelTecAutomation/CubeCell-Arduino #4: ccs811 library

raystream, 2019-10-22:
when i use the example for the ccs811 i only get the following values:

CO2: 400 TVOC: 0

solotaker, 2019-10-23:
Hello, I also found this problem,
And I found that the readings returned to normal after multiple sets of data.
I took a close look at the problem, which seems to be related to the initialization of the CSS811 sensor and seems to require a longer initialization wait time.
After testing, I found that **every time I wake up** the CSS811 again, I need to wait 5 seconds for the reading to be normal.
After that I will do more tests, thank you for your corrections.

raystream, 2019-10-23:
the problem is that after each reset the ccs811 will calibrate itself.
to get consistent data the baseline has to be set in the library.

to do this the sensor has to be running for about 20minutes and then the baseline must be exported this baseline can be set in the library.

raystream, 2019-11-01:
the set and get baseline functions are now in the CCS811 library.
ToDo:
  Find out how to store the baseline in a nonvolatile memory (maybe the sflash).

raystream, 2019-11-06:
now the baseline is stored in the userflash
see MJMCU-8128 example

closed 2019-11-06 by raystream