HelTecAutomation/CubeCell-Arduino #5: BMP180 library

raystream, 2019-10-22:
there are some errors in example for the bmp180 library.
after fixing the example i cannot compile it:

`C:\Users\Lenovo\AppData\Local\Temp\arduino_build_875617/..\arduino_cache_106657\core\core_4213195de2b82463cee8ee03e6a3b921.a(asr_board.c.o): In function `rint':

C:\Users\Lenovo\AppData\Local\Arduino15\packages\CubeCell\hardware\CubeCell\0.0.3\cores\asr650x\board\src/asr_board.c:415: multiple definition of `rint'

c:/users/lenovo/appdata/local/arduino15/packages/cubecell/tools/gcc-arm-none-eabi/8-2019-q3/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/lib/armv6-m\libm.a(lib_a-s_rint.o):s_rint.c:(.text.rint+0x0): first defined here

collect2.exe: error: ld returned 1 exit status

Mehrere Bibliotheken wurden für "BMP180.h" gefunden
 Benutzt: C:\Users\Lenovo\AppData\Local\Arduino15\packages\CubeCell\hardware\CubeCell\0.0.3\libraries\BMP180
Mehrere Bibliotheken wurden für "Adafruit_NeoPixel.h" gefunden
 Benutzt: C:\Users\Lenovo\AppData\Local\Arduino15\packages\CubeCell\hardware\CubeCell\0.0.3\libraries\RGB
Mehrere Bibliotheken wurden für "LoRaWan_APP.h" gefunden
 Benutzt: C:\Users\Lenovo\AppData\Local\Arduino15\packages\CubeCell\hardware\CubeCell\0.0.3\libraries\LoRa
exit status 1
Fehler beim Kompilieren für das Board CubeCell-Capsule.`

raystream, 2019-10-22:
after i commented "double rint(double x)" function out in asr_board.c the example compiles.
see my pull request.

solotaker, 2019-10-23:
I seem to have missed this question, it is perfect, thank you for your help.
best regards.

closed 2019-10-23 by raystream