HelTecAutomation/CubeCell-Arduino #7: GPIO7 on CubeCell

Skipper-is, 2019-10-24:
GPIO7, on the pinout says that it is battery test control. Is this capable of giving a reading on the battery level? Or is it for testing whether it is running on battery power?

Heltec-Aaron-Lee, 2019-10-28:
GPIO7 is ADC enabler, before measure the battery voltage, need set GPIO7 LOW firstly.

You can refer to this example: [https://github.com/HelTecAutomation/ASR650x-Arduino/blob/master/libraries/Basics/examples/ADC_battery/ADC_battery.ino](https://github.com/HelTecAutomation/ASR650x-Arduino/blob/master/libraries/Basics/examples/ADC_battery/ADC_battery.ino)

closed 2019-10-28 by Heltec-Aaron-Lee

tomershechtman, 2021-06-16:
Hi, page not found,
I also trying to understand how to connect battery to ADC (pullups or something) and also how to use GPIO7, currently its my interrupt pin, can I use other pins for interrupts?
Thanks