HelTecAutomation/CubeCell-Arduino #8: please add library for BMP280

raystream, 2019-10-24:
Hello,

i have added a library for the BMP280 sensor and an example for a MJMCU-8128 Sensor board.
This sensor board include an HDC1080, CCS811 and BMP280

Heltec-Aaron-Lee, 2019-10-28:
examples with the sensors you are using already here: [https://github.com/HelTecAutomation/ASR650x-Arduino/tree/master/libraries](https://github.com/HelTecAutomation/ASR650x-Arduino/tree/master/libraries)

[image]

We don't have a BMP280 sensor example now. But you can refer to the BMP180 example.

raystream, 2019-10-28:
I know that is why i have added a bmp280 library and an example in my fork

Heltec-Aaron-Lee, 2019-10-28:
Thanks for your work. I added.

closed 2019-10-28 by Heltec-Aaron-Lee