HelTecAutomation/CubeCell-Arduino #10: OneWire/DS18B20 library

Skipper-is, 2019-10-28:
Would be really great to have a library for the DS18B20 family of chips. I have tried to port the existing Arduino library, but haven't had any joy..

raystream, 2019-10-28:
hi 
i have added a OneWire library to my fork.
it compiles but can not test it ore because i have sensors

Heltec-Aaron-Lee, 2019-10-28:
Thank you for your enthusiasm @raystream :D

I will test the library if I can find a 18B20 sensor.

Skipper-is, 2019-10-28:
I've got a whole box of them on my desk, I'll give it a whirl soon (am on training for the next 3 days, but after that! It if I find some time, before that!)

Skipper-is, 2019-10-28:
Through testing yesterday, I could get the ready signal back from the device, and transmit the search command (0xF0? I can't remember now...) But there was no response, I suspect it was in the switching from output to input rapidly, it was giving problems

raystream, 2019-10-28:
i will get myself an DS18B20  and make some tests

Skipper-is, 2019-10-30:
Have you added the ASR650 to the OneWire_direct_gpio.h file?

Skipper-is, 2019-10-30:
[image]
No joy with the library I'm afraid. It is doing a bit of a search (F0), but there are no responses, I wonder whether the device isn't switching to input, and is therefore over-riding the response from the DS18B20

joergkeller, 2019-11-18:
I've got the OneWire/DallasTemperature libraries compiled, but with the same result: There is no sensor listed after search. The same wiring (with pullup resistor) and the same libraries worked with an Uno and Dragino Mini board.
@Skipper-is you asked about the OneWire_direct_gpio.h file. How/what did you enter there? The compiler currently does not define any architecture or board settings, so the library uses the default part.

joergkeller, 2019-11-18:
Just retested with the provided OneWire library and DS18x20_Temperature example in the master branch - no sensor found.

Skipper-is, 2019-11-18:
I didn't. That is partially what tripped me up. The Uno (and I am assuming all microcontrollers) have an internal port register. On the Uno, GPIO pins 0-7 are on the PORTD register, and you can use a bitmask, and bitwise operators to specify read/write modes, and Hi/Low settings. The advantage of this is to cycle signals really rapidly. 
I assume it is possible to do without the register, but I don't know how... For that we'd need to add something to the direct_gpio.h file I suspect. But that is beyond my ability!

joergkeller, 2019-11-18:
That is also my understanding. The fallback already uses digitalRead/write/pinMode, but that is probably much slower.

joergkeller, 2019-11-24:
I think I've found the solution. There are macros provided to directly modify the GPIO pins (CY_SYS_PINS_READ_PIN/CY_SYS_PINS_CLEAR_PIN/CY_SYS_PINS_SET_PIN/CY_SYS_PINS_SET_DRIVE_MODE). With some pointer arithmetic the OneWire library can directly use these.

Side note: Why is (a copy of) OneWire included in that project? Here it comes handy to quickly fix the issue but why not rely on the original OneWire library to be directly used by projects?

raystream, 2019-11-24:
That is why the library is copied, to adopt it for using with the ASR6501 ;-)

Do you have a running version?
Would be great if you could make a pull request with the needed changes.

raystream, 2019-11-25:
just tested it and it works like a charm.

thank you @joergkeller

closed 2019-11-25 by raystream

Skipper-is, 2019-11-27:
Can confirm, it is working:
`18:40:28.362 -> ROM = 28 4D 43 79 97 5 3 F2
18:40:28.395 ->   Chip = DS18B20
18:40:29.447 ->   Data = 1 3D 1 55 5 7F A5 81 66 8  CRC=8
18:40:29.481 ->   Temperature = 19.81 Celsius, 67.66 Fahrenheit
`
Great work!