HelTecAutomation/CubeCell-Arduino #11: Support for platformio coming?

prgazevedo, 2019-10-28:
Hello,

Are you planning to add support to the board in platformio?
https://docs.platformio.org/en/latest/platforms/creating_board.html

Thanks

Skipper-is, 2019-11-16:
I'd like to see it added to PlatformIO, always nice to have other IDE options!

raystream, 2019-11-16:
i use vscode without problems.
its much better than the arduino ide.

and you can use sloeber.

bertrik, 2019-12-01:
I'm also very much a platformio fan, I am happy to test support / help with getting it supported in platformio.

raystream, 2019-12-01:
there is allready an issue opened in the platform.io github repository:

https://github.com/platformio/platformio-core/issues/3078

lnlp, 2020-03-06:
It seems that work on this will now actually be started.
Looking forward to its availability. 🙂

wasn-eu, 2020-05-07:
https://github.com/HelTecAutomation/platform-asrmicro

lnlp, 2020-05-07:
'Empty repository created.'
That's sort of an update but without much information.

bertrik, 2020-06-07:
It works for me now, see https://github.com/platformio/platformio-core/issues/3078
I have a basic (empty) platformio project at
https://github.com/bertrik/cubecelldemo/blob/master/cubecelldemo.ino

It compiles, uploads and runs with
  pio run -t upload

This ticket can be closed I think

lnlp, 2020-06-07:
And any related possible issues can now be reported in its own repository: https://github.com/HelTecAutomation/platform-asrmicro650x

Heltec-Aaron-Lee, 2020-07-11:
https://github.com/HelTecAutomation/platform-asrmicro650x

closed 2020-07-11 by Heltec-Aaron-Lee