HelTecAutomation/CubeCell-Arduino #15: Libraries do not work or intergrated in to LoRaWAN Sketches so they can be used

Securethingsuk, 2019-11-02:
Hi Arron 
First let me say great hardware product. I am raising a few issues regarding software for the CubeCell.

First the major issue, I have is that the product in aimed at using LoraWAN but you have added lots of scripts for Sensors that are not related to LoRaWAN and so are of no use as they are. 

I have tried to use for example the BME680 sensor code mixed with the basic LoRaWAN example but this will not compile with errors coming back, or the sensor not found. Most of the scripts in the LoRaWAN Library here do not work. 
For example, BMP180, the only one I could get working was for HDC1080 and basic LoRaWAN sketch. My comments are:
1.	All the sensor sketches that you have put in to the library need to have them integrated into the LoRaWAN library, as that is how they are going to be used. 
a.	At least have a common way to integrate the sensors in to the LoRaWAN sketch. This will mean people can use your products with the various sensor. 
b.	I want to BME680 to send data via LoRaWAN but your sketch does not work It does not turn on Vext and does not work with LoRaWAN. It would be nice if that could include an option Line to send data In Cayenne LLP format, I can do that but the key is getting the Sensors to work with LoRaWAN. 
c.	I can get sensor to generally work with other platforms the only advantage of CubeCell is The LoRaWAN transmissions. 
2.	That all the Libraries currently are just copies from others repositories and not configured for your boards. 
a.	At least  the sketches should allow to test sensors even if not integrated with LoRaWAN
b.	The sketch needs to deal with Vext being turn on and off in the sketch and as said in an earlier issue that there needs to be a delay after the Vext is turn on.
3.	That the LoRaWAN_BMP180 sketch does not work due to Rint errors and Raystream has amended the ASR_Board .c to remove Rint. This issue may also affect other Sensors as well but I have yet to get the one I am interested in, BME680 to work.  The BMP180 works but only if the file is located in the CubeCell Board file in my case “C:\Users\User\AppData\Local\Arduino15\packages\CubeCell\hardware\CubeCell\0.0.3\cores\asr650x\board\src” this amended file needs to be put in to a new board manage file 0.0.4 so I can be easily updated by most people.

I hope this helps as I love the boards, but if no supporting basic scripts they will not sell or get used.

Simon

raystream, 2019-11-02:
I am responsible for some of the libraries.
So sorry when not working.
I Only have testet the libraries for hdc1080, ccs811 and bmp280 and these do work.
I have adopted them to work and added the baseline function for the ccs811.
I have added an example for these sensors.

I am working on the bme680 library right now (i have got the sensor today).

And i will be working on the OneWire library when my sensor ariva next week.

But beide this i think HelTec should seliger working libraries for the Commonwealth Sensors. Working libraries are need to ger These boards Sold.

Heltec-Aaron-Lee, 2019-11-06:
@Securethingsuk Thank you for your provided so many good ideas.

We added some basic sensor example just hope those libraries will help users debug. We had added HDC1080, BH1750, CCS811B, MPU9250, BMP180 those five sensors at first release, and they are worked well both with sensor basic test or LoRaWAN. thanks @raystream added some, and those examples are working well now.

V0.0.4 release will come soon, maybe this week, we added some new features in the new version, it will be a big update.

BTW. Cayenne LLP is an awesome idea!!! We are considering :D

raystream, 2019-11-14:
if uploaded a new BME680 library. this now fully works with LoRaWan
i think Aaron did a good job in reorganizing the library structure.

Heltec-Aaron-Lee, 2019-11-18:
@raystream For some third party libraries, you walked 99 steps, I just left the rest one of the way. 😄

raystream, 2019-11-22:
I think this issue can be closed.
All libraries now have a working example for standalone and lora use.

There is only the lack of the cayenne implementation.
Perhabs open a new issue for that.

Securethingsuk, 2019-11-22:
Hi yes agree this can be closed.. Cayenne can be a new implemention . I have done some before  but time limited at present I will start with some of the enviromental sensors and also the MPU 9250, as I have project for that at present , and an ideal cantidate for the CubeCell Capsule.

Closed .

closed 2019-11-22 by Securethingsuk