HelTecAutomation/CubeCell-Arduino #17: compile error

raystream, 2019-11-07:
Path Problem in Platform.txt not fixed

exec: "C:\\Users\\Dell\\AppData\\Local\\arduino15\\packages\\CubeCell\\hardware\\CubeCell\\0.0.3/tools/gcc-arm-none-eabi/bin/arm-none-eabi-g++": file does not exist
Fehler beim Kompilieren für das Board CubeCell-Board.

raystream, 2019-11-07:
After copying the Tools directory in the Hardware directory it works again
Had no time to look in the Platform file There must be a failure

Heltec-Aaron-Lee, 2019-11-08:
We provided two Platform.txt. Because install the development environment via board manager or source code have a different path.

- One in the root path is for the release version;
- Another in /tools path is for source code development environment when we run get.exe or get.py, the script will replace the one in the root path.

closed 2019-11-08 by raystream