HelTecAutomation/CubeCell-Arduino #19: Support for sub band choice within AU915

mutton74, 2019-11-14:
I am trying to use the cubecell with LoRaWAN on The Things Network (TTN).

At the moment the join process takes a long time and receiving messages is unreliable.

I believe that this is due to TTN using the second sub band (Channels 8-15) out of the 64 channels available and this library seems to be set to the first sub band (Channels 0-7)

`            // Channels
            // 125 kHz channels
            for( uint8_t i = 0; i < AU915_MAX_NB_CHANNELS - 8; i++ )
            {
                Channels[i].Frequency = 915200000 + i * 200000;
                Channels[i].DrRange.Value = ( DR_5 << 4 ) | DR_0;
                Channels[i].Band = 0;
            }
            // 500 kHz channels
            for( uint8_t i = AU915_MAX_NB_CHANNELS - 8; i < AU915_MAX_NB_CHANNELS; i++ )
            {
                Channels[i].Frequency = 915900000 + ( i - ( AU915_MAX_NB_CHANNELS - 8 ) ) * 1600000;
                Channels[i].DrRange.Value = ( DR_6 << 4 ) | DR_6;
                Channels[i].Band = 0;
            }`

TTN uses [these channels](https://www.thethingsnetwork.org/docs/lorawan/frequency-plans.html#au915-928)

raystream, 2019-11-15:
you can change it in the RegionAU915.c file.

`Channels[i].Frequency = 916800000 + i * 200000;`

mutton74, 2019-11-15:
OK thanks I changed that and it does seem to join and send on the correct channels but it misses lots of (unconfirmed) uplinks.

[image]

It has perfect reception  to a confirmed working gateway on TTN (11.5 SNR)

Any other ideas to get it more reliable?

raystream, 2019-11-15:
the change i provided only changed the uplink frequencies for the 125kHz channels but not for the 500kHz channels and the downlink.

raystream, 2019-11-23:
tried my best to add a new compiler option for the frequency band AU915_SB2

mutton74, 2019-11-23:
Sorry not much different - if there is anything I can do to help diagnose please let me know

This what I am seeing
[image]

raystream, 2019-11-23:
hope i have fixed the problem.

mutton74, 2019-11-23:
seeing this now
[image]

raystream, 2019-11-23:
Please post the frequencies used for the correct uplinks and downlinks in your screenshot.
Whats the dutycycle you have used?

mutton74, 2019-11-23:
15 secs duty cycle

Join attempt
  "lora": {
    "spreading_factor": 12,
    "bandwidth": 500,
    "air_time": 288768000
  },
  "coding_rate": "4/5",
  "timestamp": "2019-11-23T21:33:22.529Z",
  "frequency": 923300000

Count 6 - uplink
  "frequency": 917.8,
  "modulation": "LORA",
  "data_rate": "SF12BW125",
  "coding_rate": "4/5",

Count 10 uplink
  "frequency": 917.6,
  "modulation": "LORA",
  "data_rate": "SF12BW125",
  "coding_rate": "4/5",

Downlink frequencies (from gateway)
  "lora": {
    "spreading_factor": 12,
    "bandwidth": 500,
    "air_time": 329728000
  },
  "coding_rate": "4/5",
  "timestamp": "2019-11-23T21:36:58.628Z",
  "dev_addr": "26042457",
  "frequency": 923300000

Bwooce, 2019-11-24:
There's an issue that was fixed in LoRaMac-node 4.4.2 that probably has some bearing here. Check out [this bit](https://github.com/HelTecAutomation/ASR650x-Arduino/commit/ca009e05d3256f1c641d286e5c116a3d22a918fb#diff-7a80f66ff4a76731d18748b5fc7203a5R425) and think about how many channels are in the mask -- only 8 right? So this function is likely failing when it gets called.

I think this is the source of the problems for OTAA but I've not been able to trace the flow yet (it's also possible that the channel list wasn't sent in the join response for AU915 for LoRaWAN 1.0.2 or something, I remember reading that somewhere it was a bit of an error in the spec). It's weird that it would work with ADR of course, which is why it's a good puzzle.

It also points to an easier way of implementing the sub-band restriction e.g.

` // Enabling 2nd block of 8 channels (8-15) + channel 65
  MibRequestConfirm_t mibReq;

  uint16_t channelMask[] = { 0xFF00, 0x0000, 0x0000, 0x0000, 0x0002, 0x0000};
  mibReq.Type = MIB_CHANNELS_MASK;
  mibReq.Param.ChannelsMask = channelMask;
  LoRaMacMibSetRequestConfirm( &mibReq );`

[as well as removing that 20+ channel bit of code]

I'm testing this right now, and paring it back to discover if it's only required once (post-join) or more frequently (no pun intended). There were other regional changes in 4.4.2 that would be great to roll-forward too though, but quite an API change as well sadly.

Heltec-Aaron-Lee, 2019-11-25:
Hi guys, do not need so complicated at all. LoRaWAN is a complete protocol, and TTN has good support for it.

**Now, let's look into the LoRaWAN protocol firstly, I think this sentence is clear enough:**
[image]

**Then, let's look into our code:** (in LoRaWan_APP.cpp file)
[image]
Here are 5 arrays defined all channels of LoRaWAN protocol for AU915. In the channelsMaskTemp[0], 0x00FF means use the first 0~7 channels, they are 915.2MHz, 915.4MHz, 915.6MHz … 916.6MHz. When you set the value to 0xFF00, it means the upload channels changed to 8~15, they are 916.8MHz, 917.0MHz … 918.2MHz. More detail... see the below picture (Forgive me for my bad English writing😄):
[image]

**Then, let's look back to the TTN's uplink/downlink define for AU915:**
[https://www.thethingsnetwork.org/docs/lorawan/frequency-plans.html](https://www.thethingsnetwork.org/docs/lorawan/frequency-plans.html)
[image]

Generally speaking... we just need to change the `channelsMaskTemp[0]=0xFF00;`, the uplink frequency will be changed to TTN supported.

BTW. Also, the listening frequency in your LoRa gateway must the same as the TTN server. it's defined in your gateway's gloable_conf.json file.

Bwooce, 2019-11-25:
Hey I agree Aaron.Lee, I think the mask change is all that's required. The additional complication that I've uncovered is the AU915 regional definition which (for the 4.4.1 release) insists on there being 20 channels for AU915 (only, this was an Australia specific thing). This is wrong and was changed in 4.4.2 but so was a lot of the API.

Check out the RegionAU915ChanMaskSet, the top few lines validate there should be 20 channels or it fails to do anything. I think that's what is happening here, see [this bit of code](https://github.com/HelTecAutomation/ASR650x-Arduino/blob/635cee424d82074fc8dcc7835aad5d5279011976/cores/asr650x/kernel/protocols/lorawan/lora/mac/region/RegionAU915.c#L422)

RegionAU915ChanMaskSet is called eventually from your LoRaMacMibSetRequestConfirm call in your code, if you trace it all the way down.

raystream, 2019-11-25:
thanks Aaron for your help.

Using compile Option AU915SB2 now uses the normal AU915 files and switches the channelMaskTemp in LoRaWAN_APP

raystream, 2019-11-25:
Would be great if someone could test this fix.

mutton74, 2019-11-25:
Ill give it a try (does it include Bwooce's suggestions above?) 

- on a side note trying to update the library from within Arduino times out, download is too slow, yet downloading a zip and unpacking it is fine

mutton74, 2019-11-26:
Sorry might have got worse.

Does not join at all, just hangs trying, nothing seen on gateway, nothing in TTN Application and this is what Serial Monitor shows

[image]

With regards to my gateway setup.

This gateway has been operating flawlessly for about 3 years with hundreds of nodes operating and tested on it over that time using a mixture of LMIC and Microchip AU915 code.

Heltec-Aaron-Lee, 2019-11-26:
Can you please paste your gloable_conf.json file?

mutton74, 2019-11-26:
[global_conf.txt](https://github.com/HelTecAutomation/ASR650x-Arduino/files/3891347/global_conf.txt)

just converted to TXT so it would attach here

mutton74, 2019-11-26:
Using this installer from a TTN supporter

https://github.com/kersing/multitech-installer

mutton74, 2019-11-27:
I will test again tonight as there seems to be some issues with the TTN routers, just to rule out any TTN/application issues.

Heltec-Aaron-Lee, 2019-11-27:
@mutton74 Tomorrow I will test at my side.

mutton74, 2019-11-27:
OK thanks, just to confirm, I tested it again on a TTN handler that is known to be working and same result

Heltec-Aaron-Lee, 2019-11-28:
I did many tests tonight, the gateway is the HT-M02 made by us. 
[image]

 just simply fixed the `channelsMaskTemp[0] = 0xFF00;`, it's working stable after joined.
[image]
[image]

This is my [global_conf.json](https://github.com/HelTecAutomation/lorasdk/blob/master/global_conf_AU915_TTN.json), it's almost the same with @mutton74 's

Heltec-Aaron-Lee, 2019-11-28:
But this code still has a strange problem: not always join for the first time... I printed some debug info, shown the TX data been sent out more than one time, and some frequency not for AS915.
[image]

I also tested the SB2 define written by @raystream, it seems no data send out. Can I reset the code to the point before you add AU915_SB2 (Indeed I don't know how to remove one commit 😭)? I need to look into the original source code to fix this bug.

raystream, 2019-11-28:
found a failure for using the Subband 2 and corrected it in

https://github.com/HelTecAutomation/ASR650x-Arduino/commit/07dac0dffdf576b5f9d032c4c55eda834d327085

the changes are in the compiled firmware to. just flash the firmware with the AU915SB2 setting again.
the actual firmware will be downloaded and flashed

raystream, 2019-11-28:
> But this code still has a strange problem: not always join for the first time... I printed some debug info, shown the TX data been sent out more than one time, and some frequency not for AS915.
> [image]
> 
> I also tested the SB2 define written by @raystream, it seems no data send out, but I didn't find any unsuitable. Hey, @raystream , can I reset the code to the point before you add AU915_SB2 (Indeed I don't know how to remove one commit 😭)? I need to look into the original source code to fix this bug. I think the problem is caused by some error initial config.

you can just type the following in your git bash (will make a new branch with the state before my commit):

git checkout -b before_AU915SB2 deaa4df65393d4239fc93675c9602e58be9070c5

or type the following to get to the state before my commit:

git reset --hard deaa4df65393d4239fc93675c9602e58be9070c5

to take back just that commit and hope there will be no conflicts you can try:

git revert deaa4df65393d4239fc93675c9602e58be9070c5

raystream, 2019-11-29:
@mutton74 have you found time to test the new compiled AU915SB2 version or the recent github version here?

mutton74, 2019-11-30:
I have just tested it and it attempts to jin but nothing seen on the gateway or the application

Keys checked twice to be sure

[image]

[image]

Bwooce, 2019-12-06:
Hey Aaron Lee, did you make any progress? Need any help? (I know you've had some distractions)

Heltec-Aaron-Lee, 2019-12-06:
@Bwooce Yes, we are still working on this. Found some problem in the AU915 code and trying to solve it.

Heltec-Aaron-Lee, 2019-12-10:
Hi there

I found out the reason for the AU915 sub-band, the AU915 sub-band requires at least 20 125KHz channels... but normally, a gateway have 8 channels.
[image]
[image]

But my network is too bad, I will push the code tomorrow when I'm back to my office.

mutton74, 2019-12-27:
Have you had a chance to progress this any further?

Can I test anything for you again?

raystream, 2019-12-27:
Had no time to look at it cause we are just now making our modular node ready for sale.
Hope to have time next week

Heltec-Aaron-Lee, 2020-01-02:
I think this bug had been fixed already.
[https://heltec-automation-docs.readthedocs.io/en/latest/cubecell/sub_band_usage.html](https://heltec-automation-docs.readthedocs.io/en/latest/cubecell/sub_band_usage.html)

mutton74, 2020-01-05:
No, this was the result from that suggestion - again just tested - same result

> Sorry might have got worse.
>
> Does not join at all, just hangs trying, nothing seen on gateway, nothing in TTN Application and this is what Serial Monitor shows
>
> [image]
>
> With regards to my gateway [...]

Bwooce, 2020-02-01:
Back from a bit of a break now. I've just moved from my hacked up 0.4 release of Cubecell to the HEAD and my board has gone from OTAA Joining okay to not joining (as per mutton74's results).

It just sits after Joining. It is transmitting (LED is going BLUE/~GREEN(YELLOW?)/RED ) but I'm not seeing it on my gateway. 

I'm not seeing the traffic on my gateway, and I was before, so this is a software issue somewhere. I'm going to be away from AU915 gateways for a week or so, but I'll have a dig around in the meantime.

Bwooce, 2020-02-01:
I've compared my changes I did on r0.4 with the current HEAD and they're functionally ~equivalent. So this is a new problem introduced between then and now.

Bwooce, 2020-02-01:
Resolved. 

Maybe it's just that I'm using OTAA now. Putting this function prior to LoraWAN.Join()  (I already had it prior to the subsequent LoRaWAN.Send()'s but I don't know if that's really required...hey it can't hurt).

@mutton74 mutton74 please try this? 

Use the current HEAD (not r0.0.4 release from the library manager), then add

`void setTTNBand() {`

`  Serial.println("setTTNBand");`

`  // Enabling only 2nd block of 8 channels (8-15) + channel 65`
 ` MibRequestConfirm_t mibReq;`

 ` uint16_t channelMask[] = { 0xFF00, 0x0000, 0x0000, 0x0000, 0x0002, 0x0000};`
 ` mibReq.Type = MIB_CHANNELS_MASK;`
 ` mibReq.Param.ChannelsMask = channelMask;`
 ` LoRaMacMibSetRequestConfirm( &mibReq );`

`}`
 
into your sketch and call it before LoRaWAN.Join() in the loop()? Works for me like a charm.

I think this is TTN-specific as they're only using those 8 channels (of 16? available), so I'm not sure if/how that should be wedged into the CubeCell code -- we should have an example though :-)

mutton74, 2020-02-02:
PERFECT thanks @Bwooce 

I was beginning to doubt my setup - not sure why but there wasnt much left to try

Now to use this great little board

cheers
Paul

wasn-eu, 2020-02-07:
@Bwooce 
and if you encapsulate it like this than you can have it in your code and it will only be active is you select the AU915 frequencyband:

     case DEVICE_STATE_JOIN:
      {
         #if (ACTIVE_REGION == 1)
         setSubBand2();
         #endif
         LoRaWAN.Join();
         break;
      }

and make a little function out of the code from @Bwooce :

    void setSubBand2() {
      Serial.println("setSubBand2");
      // Enabling only 2nd block of 8 channels (8-15) + channel 65
      MibRequestConfirm_t mibReq;
    
      uint16_t channelMask[] = { 0xFF00, 0x0000, 0x0000, 0x0000, 0x0002, 0x0000};
      mibReq.Type = MIB_CHANNELS_MASK;
      mibReq.Param.ChannelsMask = channelMask;
      LoRaMacMibSetRequestConfirm( &mibReq );
    }

Heltec-Aaron-Lee, 2020-02-24:
Hope this make sense:
[http://community.heltec.cn/t/use-cubecell-and-m01-connect-to-ttn-au915-sub-band/1098](http://community.heltec.cn/t/use-cubecell-and-m01-connect-to-ttn-au915-sub-band/1098)

closed 2020-02-26 by wasn-eu

mikemccauley, 2020-06-19:
My experience is that the setSubBand2 solution from @Bwooce works to let the OTAA join succeed, but subsequent uplink messages would not get to the gateway (Dragino LG308). I have to alter the global userChannelsMask like this:
uint16_t userChannelsMask[6] = { 0xFF00, 0x0000, 0x0000, 0x0000, 0x0002, 0x0000}; // AU915 subband 2

MVaras, 2021-01-22:
For anyone who need the AU915 for Chile is uint16_t userChannelsMask[6] = { 0x00FF, 0x0000, 0x0000, 0x0000, 0x0001, 0x0000}; // AU915. Thanks for all you guys, this was very helpful.

Hucksleyy91, 2023-01-03:
Can we get uint16_t userChannelsMask[6] = { 0x00FF, 0x0000, 0x0000, 0x0000, 0x0001, 0x0000};
for AU915 channels 40-47 ? as they have been updated

Hucksleyy91, 2023-01-03:
> (

Hey mate im trying to get the 6th block of 8 channels for AU915 any chance you can help change this from SUBBAND 2 to 6 

uint16_t userChannelsMask[6] = { 0xFF00, 0x0000, 0x0000, 0x0000, 0x0002, 0x0000}; // AU915 subband 2

MVaras, 2023-01-03:
Im not pretty sure but...

uint16_t userChannelsMask[6] = { 0x0000, 0x0000, 0xFF00, 0x0000, 0x0006, 0x0000}; // AU915 subband 6

MVaras, 2023-01-03:
which channels are you using for RX and time delay?

AU915 SB6
500 kHz Uplink  69 911000000
500 kHz Downlink 5 926300000

Maybe change 6 to 5, because it has 0 to 7 channels
uint16_t userChannelsMask[6] = { 0x0000, 0x0000, 0xFF00, 0x0000, 0x0005, 0x0000}; // AU915 subband 6

MVaras, 2023-01-03:
That frequencies seems odd (TX and RX), you need to change RX/TX conf (and maybe delay too)

TX on freq 924000000 Hz at DR 0
Event : Tx Done
RX on freq 925700000 Hz at DR 8
Event : Rx Done

Please check this (you have to adapt to AU915, because it is for US915) https://www.baranidesign.com/faq-articles/2019/4/23/lorawan-usa-frequencies-channels-and-sub-bands-for-iot-devices

Hucksleyy91, 2023-01-03:
this is the repo
https://github.com/hkicko/CubeCell-GPS-Helium-Mapper/blob/master/src/main.cpp#L63

"AS923_AS1" Region works as i have tested,  but only at 16dbi, and 3dbm = double power
So for 14dbm difference it is 2x2x2x2x2=32 times more power 
https://docs.google.com/document/d/1AAZjdMomfmd_gUeGoqXlw9EOO4DrDHp-FIUJCmTRQ9s/edit#