HelTecAutomation/CubeCell-Arduino #20: Lack of Low Power example sketch

Stone-Hedge, 2019-11-15:
Hi,

I can't find any documentation for handling sleep modes. Would it be possible for Heltec to publish some simple example sketches on this topic?

Another issue is that while there is a ping/pong example provided, there is no simple LoRA Send or LoRA Receive sketch in the examples. I believe that a more simple example will be needed by n00bs...  

Thanks,

Dan

muzo178, 2019-11-17:
I second the request for simple Lora Send and Lora Receive sketch examples for building simple point to point solutions.

Skipper-is, 2019-11-17:
I'd like to know if there is a test method to see what has woken the device, if I'm using both a timer and a GPIO wakeup.

Skipper-is, 2019-11-17:
And is there a clock that says how long the device has been on for? I know about millis(), but this resets after a low-power.

Heltec-Aaron-Lee, 2019-11-19:
here were two sleep and wake up examples:
- [Deep sleep and wake up via timer](https://github.com/HelTecAutomation/ASR650x-Arduino/blob/master/libraries/Basics/examples/LowPower_WakeUpByTimer/LowPower_WakeUpByTimer.ino)
- [Deep sleep and wake up via GPIO](https://github.com/HelTecAutomation/ASR650x-Arduino/blob/master/libraries/Basics/examples/LowPower_WakeUpByGPIO/LowPower_WakeUpByGPIO.ino)

Stone-Hedge, 2019-11-21:
Thanks Aaron

Heltec-Aaron-Lee, 2019-11-22:
[Send and receive example](https://github.com/HelTecAutomation/ASR650x-Arduino/blob/master/libraries/LoRa/examples/pingpong/pingpong.ino)

muzo178, 2019-11-22:
> Send and receive example

How can we ping pong between the Cubecell—using this example— and the WiFi LoRa 32 (V2)? 

I tried using [OLED_LoRa_Receiver.ino](https://github.com/HelTecAutomation/Heltec_ESP32/blob/master/examples/LoRa/OLED_LoRa_Receiver/OLED_LoRa_Receiver.ino) example, but it didn’t work? 

I gather this is what Stone-Hedge is after as well?

Stone-Hedge, 2019-11-22:
> [Send and receive example](https://github.com/HelTecAutomation/ASR650x-Arduino/blob/master/libraries/LoRa/examples/pingpong/pingpong.ino)

Aaron, this is a great sketch but I think it is too complicated for n00bs and people who are learning the basics.  Would it be possible for you to write a very simple and short sketch for send and receive, similar to what was provided for the Heltec OLED Lora V2 boards?

Stone-Hedge, 2019-11-22:
> How can we ping pong between the Cubecell—using this example— and the WiFi LoRa 32 (V2)?
>
> I tried using [...]

Actually, I am having this problem but that isn't why I created this issue thread. I'll create a new issue for that.

Skipper-is, 2019-11-22:
Going back to low power, is there a way of seeing total time since switched on (or a time that persists between sleep)? Like a RTC?

Bwooce, 2019-12-02:
Hey @Skipper-is check out TimerGetCurrentTime()  (returns millis, into a type TimerTime_t). There's also a helpful companion function "TimerTime_t TimerGetElapsedTime( TimerTime_t past )" to calculate differences easily which caters for overflow/wraparound.

I'm yet to use it in real life, but the stack uses it (for duty cycle etc) and it calls RTC functions so I'd assume it has to count correctly. I'll report back later, let me know if you have success.

OstertagM, 2019-12-21:
Hey could you make an example how to go into low power with lorawan. How save save sessions keys or counter.

Heltec-Aaron-Lee, 2019-12-22:
For the currently examples with LoRaWAN all include low power features. system will automatic into sleep mode when a frame success.

closed 2020-03-06 by Heltec-Aaron-Lee

mahesh2000, 2021-05-19:
> TimerGetElapsedTime() & TimerGetCurrentTime() examples:

```
/**
 * Print uint64_t value as Arduino Serial.print does not provide this. Recursive.
 */
void print64(uint64_t value) {
    if ( value >= 10 ) print64(value / 10);    
    Serial.print((uint16_t) (value % 10));
}

uint16_t counter = 0;
TimerTime_t curTime;
TimerTime_t prevTime; 

/**
 * Keep track of time since bootup and time since last invocation
 */
void printTheTime() {
  // TimerGetElapsedTime( TimerTime_t savedTime );
  prevTime = TimerGetElapsedTime( curTime );
  curTime = TimerGetCurrentTime();
  TimerSysTime_t LastTxSysTime = { 0 };
  LastTxSysTime = TimerGetSysTime( );
  Serial.printf("[myLoop] %d do some useful work. time: ", counter++); 
  print64(uint64_t (x)); Serial.print(", elapsed since last sleep: ");
  print64(uint64_t (y)); Serial.println("");
}

```