HelTecAutomation/CubeCell-Arduino #22: BoardGetBatteryLevel() returns 100

Skipper-is, 2019-11-16:
I assume this just not implemented yet. When running BoardGetBatteryLevel(), it always returns 100.
`uint8_t BoardGetBatteryLevel()
{
	return 100;
}`
Shouldn't this return the actual bat levels from the ADC?

raystream, 2019-11-16:
I use to get the adc value:
uint16_t BatteryVoltage = GetBatteryVoltage();

BoardGetBatteryLevel() is not implemented but should return a blue of 0-100%

Skipper-is, 2019-11-17:
I just get `'GetBatteryVoltage' was not declared in this scope` when I use GetBatteryVoltage();

raystream, 2019-11-17:
are you using the git version or the release version 0.0.3?
i am using the git version and here GetBatteryVoltage() works.

Skipper-is, 2019-11-17:
I'm using the release version

raystream, 2019-11-17:
the release version is way to old.
the GetBatteryVoltage was introduced later.

Skipper-is, 2019-11-17:
Shouldn't there be a new release pushed?

raystream, 2019-11-17:
perhabs next week.

i just added a simple calculation of BatteryLevel

raystream, 2019-11-18:
i think this issue can be closed.
the issue is resolved in the recent github version.

closed 2019-11-18 by raystream

Skipper-is, 2019-11-27:
Just a question on the GetBatteryVoltage() method: What is the returned value? mV? Or a ratio?

raystream, 2019-11-28:
The blau is in mV