HelTecAutomation/CubeCell-Arduino #25: Missing ARDUINO definition

joergkeller, 2019-11-17:
Several libraries (HX711, OneWire, DallasTemperature) check the installed Arduino version before including Arduino.h:
````
#if ARDUINO >= 100
#include <Arduino.h>       // for delayMicroseconds, digitalPinToBitMask, etc
#else
#include "WProgram.h"      // for delayMicroseconds
#include "pins_arduino.h"  // for digitalPinToBitMask, etc
#endif
````

I did not find out where this definition is set, but it seems to be missing when I compile for the CubeCell board. I know that it is the version of the Arduino (IDE or compiler?) and has nothing to do with CubeCell, but it might be related with you tooling.

joergkeller, 2019-11-18:
I've also tested with the current master version, same result. This relates to a [pull request](https://github.com/HelTecAutomation/ASR650x-Arduino/pull/27).

raystream, 2019-11-18:
hi

i have merged your pull request.

closed 2019-11-18 by raystream