HelTecAutomation/CubeCell-Arduino #26: Missing ARDUINO_ARCH_XXX, LSBFIRST, MSBFIRST definitions, shift functions

joergkeller, 2019-11-17:
Using the HX711 library (version 0.7.2, https://github.com/bogde/HX711) created a compile error because some definitions and the methods shiftOut()/shiftIn() are missing in Arduino.h.

Adding this code to Arduino.h fixed the problem:
````
#define LSBFIRST 0
#define MSBFIRST 1
#define ARDUINO_ARCH_SAMD 1
````
I'm aware that for the ARDUINO_ARCH_XX definition the Arduino.h is probably the wrong place. The missing functions are probably used by many more libraries and sketches.

raystream, 2019-11-18:
added the LSBFIRST and MSBFIRST in https://github.com/HelTecAutomation/ASR650x-Arduino/commit/40ce975b185b1981184db427c349b56973502de5

added the ARCH in https://github.com/HelTecAutomation/ASR650x-Arduino/commit/546bd6289ddae0c20a89e42e3ef87fbab6d3c037

closed 2019-11-18 by raystream

joergkeller, 2019-11-18:
Being the one that triggered the ARCH change, I probably shouldn't complain...
But there are two questions from my side (even though your change did fix my immediate problem with the library):
1) Is SAMD an appropriate description of the underlying architecture of your ASR650x system? If so, I'm all fine with it, since many libraries (including HX711) check for that architecture before doing something the ARM cortex Mx way. 
2) Is Arduino.h the right place to define it, or should it go into compile settings (e.g. platform.txt)?