HelTecAutomation/CubeCell-Arduino #28: Unable to get Cubecell & Heltec Lora OLED v2 to communicate.

Stone-Hedge, 2019-11-22:
Hi,

I have some Heltec Lora OLED V1 and v2 boards (433Mhz) and I am unable to get my 433 MHZ Cubecell boards to talk to them despite being sure that the settings are the same.

Is there any reason why an older release board can't communicate with the newer ones and vice versa?

Many thanks,

Dan

Stone-Hedge, 2019-11-22:
I have a suspicion that this is down to the different ways that the Lora chips handle the syncword.  8 bit for SX127x and SX126x 16bit.  Hmmmm.

Stone-Hedge, 2019-11-24:
I have got this working. 

The Cubecell has a default sync word of 0x1424 which translates to the 8 bit sync word 0x12 on the older boards.

I'm very impressed by the Cubecell hardware but would be great to see some documentation around these syncword compatibility issues.

muzo178, 2019-11-24:
Got a link to some example code on the ESP32 Lora hardware? I briefly tried dabbling with it yesterday to no avail.. Are you setting the syncword with LoRa.setSyncWord(0x12); on the ESP32 boards?

Stone-Hedge, 2019-11-25:
> Got a link to some example code on the ESP32 Lora hardware? I briefly tried dabbling with it yesterday to no avail.. Are you setting the syncword with LoRa.setSyncWord(0x12); on the ESP32 boards?

I'd rather not share my code because it is a total spaghetti mess and will set a poor example for people wanting to improve/learn.

However, these are the settings I used on the ESP32 board to get perfect packet reception

```
  LoRa.setSpreadingFactor(7);
  LoRa.setSignalBandwidth(125E3);
LoRa.setCodingRate4(5);
  LoRa.setPreambleLength(8);
  LoRa.disableCrc();
  LoRa.setSyncWord(0x12);
```

UniquePete, 2019-12-05:
What libraries and function calls are you using in the CubeCell sketch?

My code's nothing to be proud of either, but anything helps... Here's the essence of what I use to communicate between WiFi LoRa 32 and/or Wireless Stick Lite boards (sender side):

`#include "LoRa.h"

// WIFI_LoRa_32 ports

// GPIO5 — SX1278’s SCK
// GPIO19 — SX1278’s MISO
// GPIO27 — SX1278’s MOSI
// GPIO18 — SX1278’s CS
// GPIO14 — SX1278’s RESET
// GPIO26 — SX1278’s IRQ(Interrupt Request)

#define SS 18
#define RST 14
#define DI0 26

#define BAND 917E6

bool PABOOST = true;

<other stuff>

LoRa.setPins(SS,RST,DI0);
if (!LoRa.begin(BAND, PABOOST)) {
  while (true);
}

<other stuff>

LoRa.beginPacket();             // start packet
int totalByteCount = LGO_HeaderSize + outgoingPacket.content.byteCount;
for (int i = 0; i < totalByteCount; i++) {
  LoRa.write(outgoingPacket.packetByte[i]);
}
LoRa.endPacket();             // Finish packet and send it`

Notwithstanding the above comments about the SyncWord, what I would like to know is the equivalent LoRa library and function calls for the CubeCell to those used above for the WiFi LoRa and Wireless Stick boards. If I have the sender-side details, I'm sure I'll be able to work out the receiver part.

hwaccel, 2019-12-22:
Are you looking for a sender example for a cubecell as I included my super basic sender cubecell below, I cant get my wireless stick's to read the data but one problem at a time.  (Note this code is bad dont use it as a example)

```
#include "LoRaWan_APP.h"
#include "Arduino.h"

/*
 * set LoraWan_RGB to 1,the RGB active in loraWan
 * RGB red means sending;
 * RGB green means received done;
 */
#ifndef LoraWan_RGB
#define LoraWan_RGB 0
#endif

#define RF_FREQUENCY                                915000000 // Hz

#define TX_OUTPUT_POWER                             14        // dBm

#define LORA_BANDWIDTH                              0         // [0: 125 kHz,
                                                              //  1: 250 kHz,
                                                              //  2: 500 kHz,
                                                              //  3: Reserved]
#define LORA_SPREADING_FACTOR                       7         // [SF7..SF12]
#define LORA_CODINGRATE                             1         // [1: 4/5,
                                                              //  2: 4/6,
                                                              //  3: 4/7,
                                                              //  4: 4/8]
#define LORA_PREAMBLE_LENGTH                        8         // Same for Tx and Rx
#define LORA_SYMBOL_TIMEOUT                         0         // Symbols
#define LORA_FIX_LENGTH_PAYLOAD_ON                  false
#define LORA_IQ_INVERSION_ON                        false

#define RX_TIMEOUT_VALUE                            1000
#define BUFFER_SIZE                                 30 // Define the payload size here

char txpacket[BUFFER_SIZE];
char rxpacket[BUFFER_SIZE];

static RadioEvents_t RadioEvents;

int16_t txnumber;

int16_t RSSI,rxSize;

void setup() {
    BoardInitMcu( );
    Serial.begin(115200);

    txnumber=0;
    RSSI=0;

 

    Radio.Init( &RadioEvents );
    Radio.SetChannel( RF_FREQUENCY );
    Radio.SetTxConfig( MODEM_LORA, TX_OUTPUT_POWER, 0, LORA_BANDWIDTH,
                                   LORA_SPREADING_FACTOR, LORA_CODINGRATE,
                                   LORA_PREAMBLE_LENGTH, LORA_FIX_LENGTH_PAYLOAD_ON,
                                   true, 0, 0, LORA_IQ_INVERSION_ON, 3000 ); 

   
   }

void loop()
{

		
			delay(1000);
			txnumber++;
		    sprintf(txpacket,"%s","how long can hello tx be");
        sprintf(txpacket+strlen(txpacket),"%d",txnumber);
		   
		    RGB_ON(COLOR_SEND,0);

		    Serial.printf("\r\nsending packet \"%s\" , length %d\r\n",txpacket, strlen(txpacket));

		    Radio.Send( (uint8_t *)txpacket, strlen(txpacket) );
		   
		
}

```

UniquePete, 2020-01-23:
Thanks @hwaccel. What I'd really like is the LoRa OLED V1 (or V2) code that would play pingpong with the CubeCell, the same way that the two CubeCells play with each other.

UniquePete, 2020-01-23:
Immediate problem solved. Not quite pingpong, but two way communications nonetheless. See [Ping Pong test between CubeCell dev board and WIFI LoRa 32 (V2)](http://community.heltec.cn/t/ping-pong-test-between-cubecell-dev-board-and-wifi-lora-32-v2/508/12).

closed 2020-05-07 by wasn-eu

benoitfl, 2020-11-20:
@UniquePete  I'm running into the same issue as here. But looks like the link is dead would you share an copy of what you had ? thanks

UniquePete, 2020-11-20:
Is this (http://digitalconcepts.net.au/arduino/index.php?op=gateway <http://digitalconcepts.net.au/arduino/index.php?op=gateway>) what you’re after? All my code is there. It’s not great code, but it works, and I will get around to tidying it up one day, but the essence of what is there won’t change. If you need more specific directions, let me know.

Pete