HelTecAutomation/CubeCell-Arduino #29: HTCC-AB01.pdf has GPIO pins 4/5 transposed

Bwooce, 2019-11-24:
HTCC-AB01.pdf shows GPIO5 beside RST, the silkscreen for the dev board has [4] there, and I've proved via testing that the silkscreen is correct. PDF needs correction.

Heltec-Aaron-Lee, 2019-11-25:
Yes, Thank you for your reminder, I had fixed that error.

closed 2019-11-25 by Heltec-Aaron-Lee

jagombar, 2020-04-29:
This seems to have been broken again in the latest update to the PDF

Heltec-Aaron-Lee, 2020-05-04:
Hi, @jagombar 
I check just now, but I can't find what's the problem? Can you please show me?

jagombar, 2020-05-04:
Looking at file PinoutDiagram/HTCC-AB01.pdf
The latest version (28th Jan) has GPIO 5 next to RST (incorrect I believe)
The previous version (25th Nov) has GPIO 4 next to RST (matching the PCB and correct)
Earlier versions had GPIO 5 next to RST (incorect too).
The version available at https://heltec.org/project/htcc-ab01/ is also incorrect

Heltec-Aaron-Lee, 2020-05-05:
Oh, thank you for your feedback, fixed now.