HelTecAutomation/CubeCell-Arduino #32: IfSkipJoin (stored OTAA credentials) not working

Bwooce, 2019-11-27:
Hi team, should IfSkipJoin work yet? 

I've got the call in there, but CheckNetInfo() always fails after a reboot even after a successful join (which should be storing the parameters, right?)

I have Net_Reservation set ON too (is this related? there's no docs for it yet).

Looks like great functionality.

Heltec-Aaron-Lee, 2020-01-28:
Did you try the ABP mode? ABP mode don't need join.

This part of the documents may make sense to you:
https://heltec-automation-docs.readthedocs.io/en/latest/cubecell/quick_start.html#correctly-config-the-tools-menu

Bwooce, 2020-02-01:
Thanks Aaron, it wasn't working in the release 0.4 (OTAA worked but rebooting never reused the old keys - CheckNetInfo always failed) but I'll try it now once I get AU915 working and report back.

Heltec-Aaron-Lee, 2020-02-06:
What frequency did you try?

OstertagM, 2020-02-19:
NET Reservation is NOT WORKING:
```
AT Rev 0.6

+LORAWAN=1

+KeepNet=0
+OTAA=1
+Class=A
+ADR=1
+IsTxConfirmed=0
+AppPort=2
+DutyCycle=86400000
+ConfirmedNbTrials=8

LoRaWAN EU868 Class A start!

joining...TX on freq 868100000 Hz at DR 5
Event : Tx Done
RX on freq 868100000 Hz at DR 5
Event : Rx Done
joined
Sending dev status packet
Temperature: 0.21 *C
%RH 0.34
Bat4228
unconfirmed uplink sending ...
TX on freq 867300000 Hz at DR 5
Event : Tx Done
RX on freq 867300000 Hz at DR 5
Event : Rx Timeout
RX on freq 869525000 Hz at DR 3
Event : Rx Timeout
```
[image]
You can see it on second join after reset!

Heltec-Aaron-Lee, 2020-02-21:
Seems the sending process is not successful at your side. the LoRaWAN OTAA parameters and keys are referred to the downlink message from the gateway and server. Try it at least one success sending.

Bwooce, 2020-02-27:
Answered in Issue #67 - needs a proper downlink message to be received.

closed 2020-02-27 by Bwooce