HelTecAutomation/CubeCell-Arduino #33: LoRa while off battery

Skipper-is, 2019-11-28:
I'm running the CubeCell board on a standard LiPo 18650 battery.
 I cannot get LoRa to transmit properly when running on LiPo. 
I am using the Ping Pong example, and when on USB power, it is working fine, but I unplug the USB, and power it by LiPo, and there is nothing. 
It is drawing power, and if I run another sketch (that just does the TX every 10s, followed by low power mode) it draws sensible amounts of current, followed by very low current...
Is there a GPIO I need to pull high or low to enable the Radio when on battery?

Stone-Hedge, 2019-11-29:
There shouldn't be.  

Can you confirm that the 18650 is fully charged and outputting a decent voltage? I am operating cubecell boards using 18650 cells with no issues.

Skipper-is, 2019-11-29:
It is currently running at 3.76v.

Heltec-Aaron-Lee, 2019-12-05:
Yes, it shouldn't be...

We had tested each of the SH1.25 sockets before they were sent out. Is it possible that the positive and negative terminals of the battery are reversed?

closed 2020-05-07 by wasn-eu