HelTecAutomation/CubeCell-Arduino #34: Warning regarding cubecell capsule - Actions needed

Securethingsuk, 2019-12-02:
Info As post by me on forum.This is for HEltec to action, as no Heltec action on forum., While useful comment and his guide on WASN GitHub site corrections of errors need here and action the lack of USB board and advice to  Distributors not to send out boards with out USB programming board..

Sivaelid   Sent 2 days ago

I have written this warning as BEFORE anyone uses a CubeCell capsule. They should be aware that the GITHUB and Heltec Site Diagram of the PINOUT currently does not show clearly the correct orientation . So the sensor can be placed wrongly if you look at the actual capsule and the diagram. The diagrams should have a White spot which is next to the RST button to indicated the correct orientation of the connection block.

For information my Capsule has come with:

- SENSOR INCORRECTLY FITTED . I had to seek help from GipsyBlue for his connection on his sensors. Not sure what damage that has done.

- MY CAPSULE HAD BATTERY FITTED AND WAS TOTALLY FLAT . Not good for any battery. Very large current draw when 5v added to “vs” connection to try to recover it.

- NO USB INTERFACE BOARD SUPPLIED – so cannot program

HELTEC ACTION :

Need to urgently amend drawing on this site and HEltec Site, so white dot shown and clearly mark where sensors are to be  are connected to prevent errors. As in my case sensor put in wrong at production of Distributor.

Make sure that all capsule orders have a USB interface board to program them and make clear to dealers and distributor’s this fact and make clear in sale material that the unit is supplied with board to program.

Batteries in products while good, need to be able to not run flat and discharge beyond the safe point. Add a system to disable battery or allow the user to plug in battery when they configure.
For me replace the unit and sensor with one that is working and can be programmed.

Heltec make good product like the CubeCell Board but the above lets down a good product and I would not buy again if it comes in this state.

Simon
 See picture on Heltec Forum

raystream, 2019-12-02:
for all reading this here and not on the forum.
Have a look at my documentation. I think it makes all a little bit clearer:

https://github.com/raystream/CubeCell_Getting_Started

Heltec-Aaron-Lee, 2019-12-05:
@Securethingsuk 

This afternoon, we held an urgent meeting to discuss your feedback. We worked hard to try to make things perfect, but it's obviously not enough, many things we didn't expect and not finish on time. With you guys support and feedback, we will correct everything and provide perfect documents within the next two weeks.

Apologize for our not enough.

closed 2020-05-07 by wasn-eu