HelTecAutomation/CubeCell-Arduino #35: Vext has the wrong voltage after LowPower_Handler() and wakeup

Bwooce, 2019-12-05:
After going to sleep and waking up, Vext is off. That's probably expected. Turning it back on results in ~1.6-1.8v on the pin, not the expected 3.3v and this won't properly drive most devices.

[See this community discussion,](http://community.heltec.cn/t/lowpower-handler-function-on-the-cubecell-board/582/6) I've also experienced the same thing independently.

Heltec-Aaron-Lee, 2019-12-21:
You had connect a sensor, right? That's because the current from the GPIO is inverted to Vext, which generates a certain voltage.
Please set the sensor's relevant GPIO to ANALOG while sleep mode.
for example:
`pinMode(GPIO0, ANALOG);`

lnlp, 2020-01-08:
@Bwooce Do you mean you measured 1.6 to 1.8V on the Vext pin or on GPIO6?

@Heltec-Aaron-Lee If this is about GPIO6 then the Arduino Core should properly handle this before entering deepsleep, not the user sketch.
Can you please give more details about what you exactly mean with "GPIO is inverted to Vext" and "which generates a certain voltage"?
What difference would it make (technically) to set (before entering sleep) a GPIO to ANALOG  during sleep?

eiten, 2020-07-29:
@lnlp it is common that ARM processors have the lowest power consumption of GPIO pins when they are configured as analog in. Itŝ the same with the STM32.

closed 2020-10-09 by Heltec-Aaron-Lee