HelTecAutomation/CubeCell-Arduino #37: how to receive data

raystream, 2019-12-05:
how can the CubeCell device receive downlinks and react to it (switching a relay for example)?

Heltec-Aaron-Lee, 2019-12-21:
Actually, CubeCell will print downlink message directly via the serial port.

Bwooce, 2019-12-23:
Hey Aaron Lee, it would be excellent to let this be processed programmatically. e.g. I'd like to be able to control the polling interval of my sensors by sending a downlink with the new intervals.

hwaccel, 2019-12-23:
I tried to help by submitting a basic Transmitter example and submitted it to the repo but I have no clue how to do a basic receive in an easy way like the examples for the wireless stick, hopefully someone will create a good example of receiving or even how to communicate over lora with the wireless sticks.

raystream, 2019-12-23:
@hwaccel 
You pull request was a good one.
Thank you for that.

We need a receive example for lorawan.
So we get the data in a variable and can use the data.

hwaccel, 2019-12-24:
Hi @raystream 

Sorry I'm not programmer by trade, with the lack of documentation from heltec just to get the tx example was hack and check but for a usable/working RX example I'm at a lose.  I'm thinking I may just give up on the cubecell modules using the heltec code and use and edited version of the radiolab repo as lora.recieve(str) works and it appears earlier to fallow. Unless we get someone from heltec to help make the code more usable or write a receiver example I think we are dead in the water. @Heltec-Aaron-Lee 

https://github.com/jgromes/RadioLib

Bwooce, 2019-12-25:
We're missing an API to get the data that's received right now, from what I can see. The function that gets the data is declared static (private) and doesn't have any hooks in it/doesn't store it. I'm not sure which way Aaron & co want to take this API, but it's not as simple as a new example.

Heltec-Aaron-Lee, 2019-12-26:
@raystream 
Just added a downlink example.
https://github.com/HelTecAutomation/ASR650x-Arduino/blob/master/libraries/LoRa/examples/LoRaWan_downlinkdatahandle/LoRaWan_downlinkdatahandle.ino

raystream, 2019-12-26:
great thank you very much

closed 2019-12-26 by raystream