HelTecAutomation/CubeCell-Arduino #38: DS18x20_Temperature OneWire example problem

Stone-Hedge, 2019-12-16:
Hi @raystream and thank you for publishing the examples.

When I try to compile the DS18x20 example I get the following message.

`exit status 1
Error compiling for board CubeCell-Module.`

Can you please tell me what I am doing wrong?

raystream, 2019-12-20:
you get this error because the wrong library is used.
you can see that this library is used:
  C:\Users\xxx\Documents\Arduino\libraries\OneWire/OneWire.h
the right one is:
  C:\Users\xxx\AppData\Local\Arduino15\packages\CubeCell\hardware\CubeCell\0.0.4\libraries\OneWire

just delete the wrong library (C:\Users\xxx\Documents\Arduino\libraries\OneWire)

raystream, 2019-12-20:
there is a general problem with the OneWire library with the 0.4.0 version.
fixing this is on my agenda but will take 1-2 weeks.

mikenz, 2020-01-05:
Previous PRs fixed the compilation issue with DallasTemperature.h and https://github.com/HelTecAutomation/ASR650x-Arduino/pull/50 adds DallasTemperature examples.

raystream, 2020-01-06:
thanks to @mikenz OneWire is now functional again.
Please use the GIT repository for your arduino IDE and not the one from the board manager

closed 2020-01-06 by raystream