HelTecAutomation/CubeCell-Arduino #44: No Payload will be transmitted in ABP also confirmed is off BUT not working!

OstertagM, 2019-12-29:
My configuration is:
```
Copyright @ 2019 Heltec Automation.All rights reserved.

AT Rev 0.5

+LORAWAN=1

+KeepNet=0
+OTAA=0
+Class=A
+ADR=1
+IsTxConfirmed=0
+AppPort=0
+DutyCycle=15000
+ConfirmedNbTrials=1

```
But no payload is sended:
[image]
[image]

wasn-eu, 2020-02-13:
can confirm this and it is a problem if the gateway sends data to more than one server.

OstertagM, 2020-02-13:
OK. It seems that in AOTA it worked well.
Is it possible to send only sometimes an confirmed uplink. For checking if ttn gets the package.

wasn-eu, 2020-03-08:
chirpstack server received the packages as unconfirmeddataup
only ttn server thinks that it is a confirmed one

OstertagM, 2020-03-08:
Please correct it! I use TTN

closed 2020-10-09 by Heltec-Aaron-Lee