HelTecAutomation/CubeCell-Arduino #45: Wake Up Source? Sending not working.

OstertagM, 2019-12-29:
```
LoRaWan ClassA start! 
joining...
joined
wake up by gpio <<--- Timer was wake up
T=21.70C, RH=26.09%,BatteryVoltage:4244
unconfirmed uplink sending ...
wake up by gpio <<--- Timer was wake up
T=21.66C, RH=25.99%,BatteryVoltage:4228
unconfirmed uplink sending ...
wake up by gpio  <<--- Timer was wake up
T=21.72C, RH=33.00%,BatteryVoltage:4228
unconfirmed uplink sending ...
wake up by gpio  <<<----------- This was the wake up by external interrupt

```
This is my additional code in the sketch. This comes from the example from wake up by GPIO.
It runs Alway into the wakeup() and always seems that the user key is high?! And if I press the key I have no Lorawan transmission. 

I must clearly know which my wakeup source was! And also I must send data each wakeup

```
void wakeup()
{
  delay(100);
  if(digitalRead(GPIO7)==0)
  {
    Serial.println("wake up by gpio");    
  }
}

void setup() {
    BoardInitMcu();
    Serial.begin(115200);
    //user key wake up
    pinMode(GPIO7,INPUT);
    attachInterrupt(GPIO7,wakeup,FALLING);
  
#if(AT_SUPPORT)
    Enable_AT();
#endif
    DeviceState = DEVICE_STATE_INIT;
    LoRaWAN.Ifskipjoin();
}

```

raystream, 2019-12-29:
`case DEVICE_STATE_SLEEP:
      {
        if (accelWoke) {
          if (IsLoRaMacNetworkJoined) {
            if(prepareTxFrame(APPPORT)) {
              LoRaWAN.Send();
            }
          }
          accelWoke = false;
        }
        LoRaWAN.Sleep();
        Serial.println("LoRaWAN.Sleep() finished");
        break;
      }
`
Have you make this changes?

OstertagM, 2019-12-29:
No I did not changed here anything in this state. But I can do and check it.
How can I found out the wake up cause timer or external?

raystream, 2019-12-29:
Have a look at the lora interrupt example

OstertagM, 2019-12-29:
Thanks for fast response.

closed 2019-12-29 by OstertagM

OstertagM, 2020-01-03:
That was the answer in TTN Forum

https://www.thethingsnetwork.org/forum/t/downlink-problem-join-otaa-the-things-indoor-gateway-ttig/33104/4?u=mos-fet