HelTecAutomation/CubeCell-Arduino #46: Ability to *not* use included LoRa/WAN library

lokkju, 2019-12-30:
It'd be great if the various support libraries (LoRaWAN) could be separated out from the ASR650x core; I'm wanting to use the CubeCell for Helium LongFi, which replaces the entire LoRa stack, though runs on the same SX12xx modems.

What the feasibility of partitioning this project into core board support and additional libraries, much as has been done for the Heltec ESP32 line of products?

raystream, 2019-12-30:
@lokkju 
If you dont include "LoRaWan_APP.h“ than you have only the Board without lora Support.

lokkju, 2019-12-31:
This doesn't match my experience; Just by including "Arduino.h", it seems to include headers from the `asr650x/kernel/protocols/lorawan/lora/system/*` path, specifically `timer.h`, `timeServer.h`, and `utilities.h`

sysfu, 2023-09-12:
> This doesn't match my experience; Just by including "Arduino.h", it seems to include headers from the `asr650x/kernel/protocols/lorawan/lora/system/*` path, specifically `timer.h`, `timeServer.h`, and `utilities.h`

Did you ever find a resolution to this? I'm interested in programming CubeCell hardware with only the necessary code and functions to support the Helium LoRaWAN network.

lokkju, 2023-09-12:
I did not, I went a different direction for my project, and haven't looked at the code in years.

sysfu, 2023-09-12:
> I did not, I went a different direction for my project, and haven't looked at the code in years.

Did you opt for another Arduino LoRaWAN hardware platform similar to the CubeCells? Or something more radically different?

KunYi, 2023-09-13:
maybe try this https://github.com/helium/longfi-platformio/tree/master/Heltec-CubeCell-Board