HelTecAutomation/CubeCell-Arduino #47: Questions: JOIN nonce how is it generated?

OstertagM, 2019-12-30:
I have a question: How is the join nonce generated? Is it a being  “new” after reset and not a value that was previously used or just inkremented?

Heltec-Aaron-Lee, 2020-03-06:
This document may make sense to you:
[http://www.techplayon.com/lora-device-activation-call-flow-join-procedure-using-otaa-and-abp/](http://www.techplayon.com/lora-device-activation-call-flow-join-procedure-using-otaa-and-abp/)

The join nonce will be new after reset.

OstertagM, 2020-03-06:
> The DevNonce is a counter starting from 0 when the device is initially powered up and incremented with every Join-request by End Device. A DevNonce value shall never be reused for a given AppEUI value. If the End Device can be power-cycled then DevNonce shall be persistent (stored in a non-volatile memory). Resetting DevNonce without changing AppEUI will cause the Network Server to discard the Join-requests of the device For each end-device, the Network Server keeps track of the last DevNonce value used by the end-device, and ignores Join-requests if DevNonce is not incremented.

How do be sure that this is unique an grated once?