HelTecAutomation/CubeCell-Arduino #48: JOIN accapt is not received by node!

OstertagM, 2019-12-31:
I have made test yesterday. All was ok. Join after one try. Today now I can't what's wrong?

[image]

Heltec-Aaron-Lee, 2020-01-02:
Are you connected to TTN's server? TTN has a limit on the number of downlinks sent every day, can this be the reason?

OstertagM, 2020-01-02:
Yes I connected it to TTN but as you can see in second picture timestamp there is no reason for this downlink limit. Because the acceptance was sended out. The picture you see is only from one node. 
I think it is an timing error or something. 
But I have no chance to Debug. 
Or can it switche some Debug to serials output on? 
This would be a good option. 
We should have the possibility to see when which state is achieved in microseconds. Or if something is received but wrong crc or something.

raystream, 2020-01-02:
what gateway are you using?
i had this issue with the ttn indoor gateway every day and needed to restart it.
With my mikrotik based gateways or the heltec gateways i never had this issue.

OstertagM, 2020-01-03:
Yes it is the TTN indoor Gateway. Because it was cheap. 
How could we get help here? I think also perhaps you can share your experience with TTN

OstertagM, 2020-02-21:
It worked now. I updated the complete lib.
I'll observe the problem.

closed 2020-02-21 by OstertagM