HelTecAutomation/CubeCell-Arduino #51: kernel/protocols/lorawan/lora/readme.md is from old version without SX126x support

lnlp, 2020-01-05:
File [kernel/protocols/lorawan/lora/readme.md](https://github.com/HelTecAutomation/ASR650x-Arduino/blob/master/cores/asr650x/kernel/protocols/lorawan/lora/readme.md) is from [LoRaMac-node v4.3.2](https://github.com/Lora-net/LoRaMac-node/tree/v4.3.2) while SX126x radio support was only added in [LoRaMac-node](https://github.com/Lora-net/LoRaMac-node) v4.4.1.

Above gives the impression that the contents of [this repository](https://github.com/HelTecAutomation/ASR650x-Arduino) is inconsistent, outdated and/or incomplete.

lnlp, 2020-03-06:
@Heltec-Aaron-Lee 
Please explain, is ASR650x-Arduino core still based on LoRaMac-node v4.3.2?

- If correct that would mean that SX1262 support (which is required for ASR650x) would be based on custom Heltec code added to LoRaMac-node v4.3.2.

    _That would imply that ASR650x-Arduino would us a Heltec customized version of LoRaMac-node where SX1262 support is added by Heltec custom code, In that case I would be worried because it would be hard to maintain, because SX1262 was added in a later LoRaMac-node version and so this would make it very difficult to quickly support new LoRaMac-node versions._

- If not correct then please update this README.md

Heltec-Aaron-Lee, 2020-03-06:
Now it's 4.4.0. I will update the readme.md now.

lnlp, 2020-03-06:
And hopefully also the documentation. 😉

Heltec-Aaron-Lee, 2020-03-06:
Indeed we got the SDK from ASR, I had looked into the ASR's SDK and original [LoRaMac-node](https://github.com/Lora-net/LoRaMac-node), I believe the LoRaWAN stack is mixed with 4.4.0 and 4.4.1

closed 2020-05-07 by wasn-eu