HelTecAutomation/CubeCell-Arduino #54: micros() function missing in Arduino.h

oscaropenness, 2020-01-14:
For using ultrasonic sensors, accurate (microsecond) time measurements are necessary.
The `micros()` function seems currently not implemented in the board definition for the ArduinoIDE.
Using the implemented `millis()` function is unfortunately not precise enough.

Can the `micros()` function be implemented in `Arduino.h` as well?

heyderpaez, 2020-01-15:
I have the same issue. I have not even been able to execute the basic examples for HTCC-AB01 due to the error of the micros () function.

lnlp, 2020-01-31:
This appears to be a bug and really should be fixed for compatibility with the basics of the Arduino framework.

OstertagM, 2020-02-19:
I have same problem

kkalbaugh, 2020-03-11:
Also having the same issue with Lorawan example and using a dallas DS18B20.

Heltec-Aaron-Lee, 2020-03-16:
Just added two examples: [micros](https://github.com/HelTecAutomation/ASR650x-Arduino/blob/master/libraries/Basics/examples/micros/micros.ino) and [millis](https://github.com/HelTecAutomation/ASR650x-Arduino/blob/master/libraries/Basics/examples/millis/millis.ino).

**micros**
The micros function uses the internal system ticker (48MHz timer), this timer of ASR6501 is not accurate, as tested, it has 1.7% error.
[image]

The delay in the example is 1000ms, ideally, the output value here should be 1000000.

But look at the printed time stamp, each print have 20 - 22ms error. It means the actual `delay(1000); ` spend 1020 - 1022ms. Because the delay uses internal system ticker too.

So the error is `(1020000 - 1002735) / 1020000 ≈ 1.7%`

That’s why I didn’t use system ticker for LoRaWAN operations. Another problem is the system ticker can’t run during the deep sleep period, wake up from deep sleep will make the micros back to 0.

**millis**
[image]

The millis function uses external 32.768KHz RTC clock, it a very accurate timer, and can run during the deep sleep period. As the picture has shown, it’s the same with the delay time.

But the minimum period is `1/32768 ≈ 30us`, can’t be the source of micros.

Heltec-Aaron-Lee, 2020-03-16:
This problem also talk here:
[http://community.heltec.cn/t/cubecell-micros-function/985/3](http://community.heltec.cn/t/cubecell-micros-function/985/3)

closed 2020-03-17 by Heltec-Aaron-Lee