HelTecAutomation/CubeCell-Arduino #55: HTCC-AB01 pinout diagram can easily lead to errors and possible damage.

lnlp, 2020-01-15:
In the HTCC-AB01 pinout diagram pin17/GPIO7 and pin16/GPIO6 look like they are standard pins on the side of the board (situated next to the GND pins). But in reality they are not.

**This causes confusion and can easily lead to incorrect wiring which in worst case can damage the board and other components connected to it.**

In the current version of the diagram it looks like the GPIO7 and GPIO6 pins are regular pins on the side of the board because that is how they are drawn in the diagram (but in reality they are not). When looking at a glance this gives incorrect information to the user. Only when looking very carefully one can see that GPIO7 and GPIO6 are actually not on the side of the board which is is therefore easily overlooked and error prone.

Why is this a problem:
The pins are not labeled on the top side (only on the bottom side). When using the HTCC-AB01 on a breadboard one has to look at the pinout diagram and then count the pin sequence numbers to know which pin is what. Looking at the pinout diagram one can easily overlook that GPIO7 and GPIO6 are not actual pins because they are represented as actual pins. Therefore one can easily count bad sequence numbers and as a result wire the ATCC-AB01 incorrectly. This is not only impractical but may also lead to damage.

**Therefore please change the pinout diagram and draw the information for GPIO7 and GPIO6 at another area in the diagram, not next to and not in row with the side pins.
It visibly should become very clear (at a glance) that GPIO7 and GPIO6 are NOT regular pins on the side of the board.**

raystream, 2020-01-15:
There are red lines connecting the label to the position of the pins.
I dont think anyone would not see them

lnlp, 2020-01-15:
Maybe you should think again (and harder).

I provided above feedback on my own experience and common sense and I did spent quite some efforts explaining why this is an issue.

If you care about your users and want to prevent them from making accidental mistakes (and destroying their hardware), then it should not be too much efforts to update the diagram _(which should be simple when you have the source)_.

About 4.5% of the world population is color blind. Many/most of them will not notice the red line at all. But the red lines are not enough to correct the optical effect that pins GPIO7 and GPIO6 are actually drawn like all the other pins, looking like they are on the side.

mikenz, 2020-01-16:
I agree with @lnlp on this one, the first time I went to use the board on a breadboard I started counting the pins off from the usb socket end and was off by one too.

I would suggest something like this:
[image]

lnlp, 2020-01-16:
> I would suggest something like this:

That is exactly what I had in mind.

Additionally I'd like to suggest to use available space more efficiently (use less white space). 
I usually print my pinout diagrams and then laminate them for easy use.
Below example is optimized for printing on A4 paper size:

[image]

raystream, 2020-01-28:
fixed in 
https://github.com/HelTecAutomation/ASR650x-Arduino/commit/7e16adb74ae18b796e1c4a6151f23c55962b7b5f

closed 2020-01-28 by raystream