HelTecAutomation/CubeCell-Arduino #57: Information in HTCC-AB01 circuit diagram (and possibly in FAQ) is incorrect

lnlp, 2020-01-15:
In FAQ article [how-to-use-the-adc-pin-for-analogread-asr6501](https://docs.heltec.cn/#/en/faq/cubecell_series_common_problem_summary?id=how-to-use-the-adc-pin-for-analogread-asr6501) the following is mentioned:

> The ASR6501 embedded 12-bit 1Msps SAR ADC, but only have one channel ADC input pin, this pin had been used for battery voltage reading default.
> 
> In order to read other analog signal, user need remove the resistance BR1, then the ADC header will be free.

The name BR1 refers to a solder bridge which it is not because in reality this is a resistor (probably 0 ohm) it is not a solder bridge and without the board [IMAGE ](https://docs.heltec.cn/en/faq/img/cubecelladcbridge.png)in the FAQ one would never know where to find it.

In the [HTCC-AB01 circuit diagram](https://github.com/HelTecAutomation/ASR650x-Arduino/blob/master/BlockDiagram/HT-AB01.pdf) this resistor is incorrectly drawn with a solder bridge symbol and named as a solder bridge (BR) instead of a resistor (R). **The circuit diagram should be corrected**.

This resistor should be labeled on the silk screen of the PCB but on the current version it is not.

On my board resistor 'BR1' is missing which conflicts with the information in the FAQ.

Is the **missing (0 Ohm) resistor** 'BR1' on my board an incident or is this a structural production change of the board?
**If structural change** then the information "this pin had been used for battery voltage reading default" in the [**FAQ** ](https://docs.heltec.cn/#/en/faq/cubecell_series_common_problem_summary?id=how-to-use-the-adc-pin-for-analogread-asr6501) is incorrect and **should be updated**.

raystream, 2020-01-27:
in the block diagram there is a 0R beneath BR1 label.
it is shown as solder bridge because it is a "switch" but the 0R indicates that it is produced as a 0Ohm resistor on the pcb for easier manufacturing.

lnlp, 2020-01-31:
It is inconsistent from whatever perspective you look at it, which is bad and confusing.
It should be fixed, one way or the other. Either in the circuit diagram or on the PCB and it should be added to the silkscreen.

Heltec-Aaron-Lee, 2020-03-17:
Hi @lnlp 

The main reason is for easier manufacturing. I will fix the release version diagram.

closed 2020-05-07 by wasn-eu