HelTecAutomation/CubeCell-Arduino #58: ASR650x-Arduino API conflicts with Arduino naming conventions

lnlp, 2020-01-15:
Unfortunately the ASR650x-Aruino API is yet undocumented.
_(Currently only AT commands are documented, but even these can be improved because essential technical details are missing.)_

From some of the code examples it becomes clear that the API uses names for functions and class methods that start with an upper-case character while the standard Arduino convention is to use function and method names that start with a lower-case character.

Using naming conventions that conflict with regular Arduino naming conventions results in (application) source code with mixed (conflicting) naming conventions.
This is a bad thing and poor design. This makes it more difficult to read and understand code but also makes it more difficult to write code, especially with an IDE like the Arduino IDE that lacks code-completion. Forcing users to use multiple conflicting naming conventions in their code makes the code messy and ugly.

**ASR6501x-Arduino is an Arduino Core. It should therefore use and conform to common Arduino naming conventions. Just like all major Arduino Cores, all official libraries and most third party Arduino libraries.**

Heltec-Aaron-Lee, 2020-01-24:
Thank you for your great suggestions, We will take your suggestions and try to correct this series of issues when the 6502 is released.

lnlp, 2020-01-31:
These corrections should be made as soon as possible because this will be a breaking change that impacts application code being written for CubeCell devices. The longer you wait the more code that customers will have to correct for the changes.
This is not dependent on a the ASR6501 or a newer ASR6502. These corrections should be made asap and be used for all ASR650x MCU's.
ASR650x is hardware produced by ASR but ASR650x-Arduino is software (Arduino Core) written by Heltec.

Heltec-Aaron-Lee, 2020-02-01:
@lnlp I had talked with my colleagues, we will begin the changes when we back to work. Because of the virus, we have to delay the time back to work.

fredolaredo, 2020-02-08:
Wish you lot of courage ! I can't imagine what is like ! By the way, I am also quite disappointed with the naming conventions, and agree on the fact that the sooner, less changes we will have to do in the code. For now , just digging into this promising platform. Thx

wasn-eu, 2020-02-26:
think that this issue is solved with the last update

closed 2020-02-26 by wasn-eu