HelTecAutomation/CubeCell-Arduino #59: Ability to use multiple sensors in a capsule

kalon33, 2020-01-27:
Dear Heltec,

It would be great to be able to have multiple sensors in a capsule, notably HDC1080 + CCS811B, as CCS811B sensors would benefit from temp and humidity corrections for CO2 and tVOC measurements accurateness, and more globally because it is useful to have multiple sensors on one node.

raystream, 2020-01-27:
you can have more than one sensor.
for example this one: [CJMCU8128b](https://www.wasn.eu/p/cubecell-capsule-sensor-board-mit-cjmcu-8128)
this sensor combines a CCS811, HDC1080 and BMP280

Heltec-Aaron-Lee, 2020-01-28:
We will have multiple sensor support in ASR6502 version.

kalon33, 2020-02-03:
> you can have more than one sensor.
> for example this one: [CJMCU8128](https://www.wasn.eu/p/cubecell-capsule-sensor-board-mit-cjmcu-8128)
> this sensor combines a CCS811, HDC1080 and BMP280

Will it be back in stock soon? How does it compete with this one for example https://www.wasn.eu/p/cubecell-capsule-sensor-board-mit-bme680-und-bh1750 on accuracy?

wasn-eu, 2020-02-13:
I am sorry for rhe late reply.
We are focusing on our modular node and other smart devices.
The capsule isnice to show what a little device can do nut not practical for the most  applications we have.

closed 2020-02-26 by wasn-eu