HelTecAutomation/CubeCell-Arduino #62: Stream.h : return value for read() should be "int" (not uint32)

fredolaredo, 2020-02-09:
compared to other Arduino packages 

./Heltec-esp32/hardware/esp32/0.0.4/cores/esp32/Stream.h
    virtual int read() = 0;
./Moteino/hardware/samd/1.3.0/cores/arduino/Stream.h
    virtual int read() = 0;
./esp32/hardware/esp32/1.0.4/cores/esp32/Stream.h
    virtual int read() = 0;
./arduino/hardware/avr/1.8.2/cores/arduino/Stream.h
    virtual int read() = 0;
./CubeCell/hardware/CubeCell/0.0.4/cores/asr650x/cores/Stream.h
    **virtual uint32 read() = 0;**
./industruino/hardware/samd/1.0.1/cores/industruino/Stream.h
    virtual int read() = 0;

fredolaredo, 2020-02-09:
same for HardwareSerial.h

Heltec-Aaron-Lee, 2020-02-15:
Thank you for your hints, I had updated it.

closed 2020-02-15 by Heltec-Aaron-Lee