HelTecAutomation/CubeCell-Arduino #66: v0.0.5 : undefined references on compilation

fredolaredo, 2020-02-12:
many compilation errors probably due to the renaming actions in 0.0.5 version, but I'm no expert
-> no problem on the same code when going back to 0.0.4

```
/Users/fred/Library/Arduino15/packages/CubeCell/hardware/CubeCell/0.0.5/libraries/LoRa/src/LoRaWan_APP.cpp:149: undefined reference to `devEui'
/Users/fred/Library/Arduino15/packages/CubeCell/tools/gcc-arm-none-eabi/8-2019-q3/bin/../lib/gcc/arm-none-eabi/8.3.1/../../../../arm-none-eabi/bin/ld: /Users/fred/Library/Arduino15/packages/CubeCell/hardware/CubeCell/0.0.5/libraries/LoRa/src/LoRaWan_APP.cpp:149: undefined reference to `appEui'
/Users/fred/Library/Arduino15/packages/CubeCell/tools/gcc-arm-none-eabi/8-2019-q3/bin/../lib/gcc/arm-none-eabi/8.3.1/../../../../arm-none-eabi/bin/ld: /Users/fred/Library/Arduino15/packages/CubeCell/hardware/CubeCell/0.0.5/libraries/LoRa/src/LoRaWan_APP.cpp:149: undefined reference to `appKey'
/Users/fred/Library/Arduino15/packages/CubeCell/hardware/CubeCell/0.0.5/libraries/LoRa/src/LoRaWan_APP.cpp:98: undefined reference to `isTxConfirmed'
/Users/fred/Library/Arduino15/packages/CubeCell/tools/gcc-arm-none-eabi/8-2019-q3/bin/../lib/gcc/arm-none-eabi/8.3.1/../../../../arm-none-eabi/bin/ld: /Users/fred/Library/Arduino15/packages/CubeCell/hardware/CubeCell/0.0.5/libraries/LoRa/src/LoRaWan_APP.cpp:98: undefined reference to `appPort'
/Users/fred/Library/Arduino15/packages/CubeCell/tools/gcc-arm-none-eabi/8-2019-q3/bin/../lib/gcc/arm-none-eabi/8.3.1/../../../../arm-none-eabi/bin/ld: /Users/fred/Library/Arduino15/packages/CubeCell/hardware/CubeCell/0.0.5/libraries/LoRa/src/LoRaWan_APP.cpp:98: undefined reference to `confirmedNbTrials'
/Users/fred/Library/Arduino15/packages/CubeCell/hardware/CubeCell/0.0.5/libraries/LoRa/src/LoRaWan_APP.cpp:453: undefined reference to `loraWanClass'
/Users/fred/Library/Arduino15/packages/CubeCell/tools/gcc-arm-none-eabi/8-2019-q3/bin/../lib/gcc/arm-none-eabi/8.3.1/../../../../arm-none-eabi/bin/ld: /Users/fred/Library/Arduino15/packages/CubeCell/hardware/CubeCell/0.0.5/libraries/LoRa/src/LoRaWan_APP.cpp:453: undefined reference to `loraWanAdr'
/Users/fred/Library/Arduino15/packages/CubeCell/hardware/CubeCell/0.0.5/libraries/LoRa/src/LoRaWan_APP.cpp:533: undefined reference to `overTheAirActivation'
/Users/fred/Library/Arduino15/packages/CubeCell/tools/gcc-arm-none-eabi/8-2019-q3/bin/../lib/gcc/arm-none-eabi/8.3.1/../../../../arm-none-eabi/bin/ld: /Users/fred/Library/Arduino15/packages/CubeCell/hardware/CubeCell/0.0.5/libraries/LoRa/src/LoRaWan_APP.cpp:533: undefined reference to `devEui'
/Users/fred/Library/Arduino15/packages/CubeCell/tools/gcc-arm-none-eabi/8-2019-q3/bin/../lib/gcc/arm-none-eabi/8.3.1/../../../../arm-none-eabi/bin/ld: /Users/fred/Library/Arduino15/packages/CubeCell/hardware/CubeCell/0.0.5/libraries/LoRa/src/LoRaWan_APP.cpp:533: undefined reference to `appEui'
/Users/fred/Library/Arduino15/packages/CubeCell/tools/gcc-arm-none-eabi/8-2019-q3/bin/../lib/gcc/arm-none-eabi/8.3.1/../../../../arm-none-eabi/bin/ld: /Users/fred/Library/Arduino15/packages/CubeCell/hardware/CubeCell/0.0.5/libraries/LoRa/src/LoRaWan_APP.cpp:533: undefined reference to `appKey'
/Users/fred/Library/Arduino15/packages/CubeCell/tools/gcc-arm-none-eabi/8-2019-q3/bin/../lib/gcc/arm-none-eabi/8.3.1/../../../../arm-none-eabi/bin/ld: /Users/fred/Library/Arduino15/packages/CubeCell/hardware/CubeCell/0.0.5/libraries/LoRa/src/LoRaWan_APP.cpp:533: undefined reference to `devAddr'
/Users/fred/Library/Arduino15/packages/CubeCell/tools/gcc-arm-none-eabi/8-2019-q3/bin/../lib/gcc/arm-none-eabi/8.3.1/../../../../arm-none-eabi/bin/ld: /Users/fred/Library/Arduino15/packages/CubeCell/hardware/CubeCell/0.0.5/libraries/LoRa/src/LoRaWan_APP.cpp:533: undefined reference to `nwkSKey'
/Users/fred/Library/Arduino15/packages/CubeCell/tools/gcc-arm-none-eabi/8-2019-q3/bin/../lib/gcc/arm-none-eabi/8.3.1/../../../../arm-none-eabi/bin/ld: /Users/fred/Library/Arduino15/packages/CubeCell/hardware/CubeCell/0.0.5/libraries/LoRa/src/LoRaWan_APP.cpp:533: undefined reference to `appSKey'
/Users/fred/Library/Arduino15/packages/CubeCell/hardware/CubeCell/0.0.5/libraries/LoRa/src/LoRaWan_APP.cpp:551: undefined reference to `loraWanClass'
/Users/fred/Dropbox/__Technologie/Arduino/libraries/ESP8266_and_ESP32_Oled_Driver_for_SSD1306_display/src/OLEDDisplay.cpp:855: undefined reference to `yield()'
D:\asr_lib/AT_Command.c:1127: undefined reference to `keepNet'
D:\asr_lib/AT_Command.c:1179: undefined reference to `keepNet'
D:\asr_lib/AT_Command.c:1260: undefined reference to `keepNet'
```

Heltec-Aaron-Lee, 2020-02-15:
In this update, we have modified the names of some variables to make them more consistent with the style of Arduino.

Which example did you use?

fredolaredo, 2020-02-15:
one of my own
[CubeCell_Timer_BMP280_SSD1306_LoRa.ino.zip](https://github.com/HelTecAutomation/ASR650x-Arduino/files/4208948/CubeCell_Timer_BMP280_SSD1306_LoRa.ino.zip)

wasn-eu, 2020-02-26:
it can be so easy.
change this line in your code

> bool KeepNet = false;

to

> bool keepNet = false;

closed 2020-02-26 by wasn-eu