HelTecAutomation/CubeCell-Arduino #67: Save and restore sessions secures after the first join

OstertagM, 2020-02-13:
How can I save the session secrets in ee prom and restore it? I don't want to do an ttn rejoin after complete loose of power.

Heltec-Aaron-Lee, 2020-02-15:
Just keep this option ON.
[image]

Heltec-Aaron-Lee, 2020-02-15:
BTW. This document may make sense to you:
[https://heltec-automation-docs.readthedocs.io/en/latest/cubecell/quick_start.html](https://heltec-automation-docs.readthedocs.io/en/latest/cubecell/quick_start.html)

[image]

OstertagM, 2020-02-15:
Thanks a lot.

OstertagM, 2020-02-15:
> Enable LORAWAN_Net_Reservation must disable Frame counter (fCnt) in LoRa server.
I think frame counts could also be safed and used again. What happens if I do not deactivate it? With what frame count starts after reset? Sorry for asking I can't test it at moment.

OstertagM, 2020-02-19:
IS NOT working see now post:
https://github.com/HelTecAutomation/ASR650x-Arduino/issues/32

Heltec-Aaron-Lee, 2020-02-21:
On the CubeCell board, there doesn't have an EEPROM, so the OTAA related parameters and keys are directly stored in FLASH. The number of flash reads and writes is often less than 100,000. If each downlink is stored, this will seriously reduce the life of the flash. So we store it every 15,000 times, which is why fCnt is turned off.

OstertagM, 2020-02-21:
OK so if I connect to an EEProm external and write read here it would work!
But why did net reservation not work?

Heltec-Aaron-Lee, 2020-02-24:
You need to make sure the node has successfully received last downlink message.

OstertagM, 2020-02-24:
I know. But als there is a gap for missing some downlink.
The important is that I can send only f.e. Each 100 an confirmed one and the other a unconfirmed is that possible?

Heltec-Aaron-Lee, 2020-02-24:
Which LoRa server did you use? Can you please upload your global_conf.json file? I think the problem is because of you gateway have a different listening frequency with CubeCell sending frequency.

OstertagM, 2020-02-24:
I use TTN and it's indoor Gateway. So I don't have this file. And what has your answer todo with my question? I want to know why the node doesn't save keys if net reserve is on? And if I can send uplink confirmed and unconfirmed in same sketch?

Heltec-Aaron-Lee, 2020-02-25:
I had just tested and confirmed at my side, the keys save at my side, there are two important keys to success:
- must have at least one success downlink message from LoRa server (Joined can't be);
- Must disable fCnt (Frame counter) in the server.

No, only confirmed or unconfirmed in the same sketch.

OstertagM, 2020-02-25:
You know where is limitation on downlink in TTN. So I can do not always sending a confirmed uplink! Please provide a possibility in your lib to switch on confirmed for one uplink!

Heltec-Aaron-Lee, 2020-02-25:
That's very easy...

when you received the first downlink message (by judge the fCnt), just set the `isTxConfirmed` to `false` by software.
[image]

closed 2020-05-07 by wasn-eu