HelTecAutomation/CubeCell-Arduino #70: RGB support

dbuggz, 2020-02-26:
I'm using the CubeCell (HTCC_AB01) and the RGB example fails as follows:

_`In file included from C:\Users\Dave\Documents\Arduino\hardware\CubeCell\ASR650x-Arduino\libraries\RGB\examples\RGB\RGB.ino:1:0:

C:\Users\Dave\Documents\Arduino\libraries\Adafruit_NeoPixel/Adafruit_NeoPixel.h: In member function 'boolean Adafruit_NeoPixel::canShow() const':

C:\Users\Dave\Documents\Arduino\libraries\Adafruit_NeoPixel/Adafruit_NeoPixel.h:234:58: error: 'micros' was not declared in this scope`_

seems that 'micros' has not been defined for the CubeCell Core

dbuggz, 2020-02-27:
This looks like issue #54. please add my request there.

Heltec-Aaron-Lee, 2020-03-06:
[image]

I can compile at my side... how did you install the framework?

dbuggz, 2020-03-15:
Aaron,

i found it was using the Adafruit library, but complied OK with your library. Why is the Cubecell library different to the Adafruit one?
Dave

Heltec-Aaron-Lee, 2020-03-16:
Hi @dbuggz
I had just added two examples, maybe that's the answer. :D
[https://github.com/HelTecAutomation/ASR650x-Arduino/issues/54#issuecomment-599449437](https://github.com/HelTecAutomation/ASR650x-Arduino/issues/54#issuecomment-599449437)

dbuggz, 2020-03-16:
Thanks Aaron

closed 2020-03-17 by Heltec-Aaron-Lee