HelTecAutomation/CubeCell-Arduino #72: Compiling error because of my Windows username

samy4sam, 2020-03-17:
I got the following error:
`exec: "C:\\Users\\.....": file does not exist
Error compiling for board CubeCell-Board.`

I tracked down this error to my Windows username, which contains a space. 
I tried my code on an other userprofile on my computer and it worked perfectly.  

Please fix this. thx

Heltec-Aaron-Lee, 2020-04-09:
Indeed I don't know how to fix this problem...

wasn-eu, 2020-05-07:
can you please test the github version with a username that has spaces?
i might have fixed that in https://github.com/HelTecAutomation/ASR650x-Arduino/commit/9aa19bc54481f11bd8c5c3d0820e784c73f7a54e

samy4sam, 2020-05-08:
Still getting an error:

`Arduino: 1.8.12 (Windows 10), Board: "CubeCell-Board, REGION_AS923, CLASS_A, OTAA, ON, CONFIRMED, OFF, ON, ACTIVE, None"

'C:\Users\firstname' is not recognized as an internal or external command,

operable program or batch file.

[14644] Failed to execute script CubeCellelftool

Traceback (most recent call last):

  File "CubeCellelftool.py", line 3, in <module>

  File "elftool.py", line 26, in elftool.main

FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Users\\firstname~1\\AppData\\Local\\Temp\\arduino_build_13677/ADC.ino.hex'

exit status -1
Fehler beim Kompilieren für das Board CubeCell-Board.`

wasn-eu, 2020-05-08:
there cant be more done in the platform.txt files.
now its time to look in the CubeCellftool but i dont have the sources to take a deeper look.

Dtrieb, 2020-09-09:
Same problem here.

alexcorvis84, 2020-09-30:
> Same problem here.

As commented on first post, if you code on another userprofile (without spaces) It will work perfectly.

I was getting the same error as well until I found the solution.

Hope this could be fixed somehow. I've checked few users have suffered already this problem.

ShadowLp174, 2022-02-01:
Same problem here :/

ShadowLp174, 2022-02-02:
Hey all,
I've got a working workaround for the problem which doesn't require changing your username or modifying the registry.
1. Create a Symlink in the C:\Users folder for your username, which doesn't contain whitespaces.
2. Locate your platform.txt file. It is usually installed in C:\Users\<YourUsername>\Documents\Arduino\hardware\CubeCell\ASR650x-Arduino
3. Copy the folder path with symlink. For example: C:\Users\<YourSymlink>\Documents\Arduino\hardware\CubeCell\ASR650x-Arduino
4. Open the platform.txt file with your favourite text editor. It should have search and replace (Editor should work)
5. Hit Ctrl+F
6. In the "search" field insert ```{runtime.platform.path}``` and in the "replace" field insert your symlink runtime path (for example: C:\Users\<YourSymlink>\Documents\Arduino\hardware\CubeCell\ASR650x-Arduino)
7. Hit "replace all"

The compiler should stop failing and the upload to the CubeCell board should work perfectly!
Have fun!