HelTecAutomation/CubeCell-Arduino #73: Battery level Measurement without external component

nidaas, 2020-03-25:
Hi, Is there  a way to measure MCU voltage without any external component.? Voltage level OE interrupt of low voltage also works in my application.
Thank You
Nidaas

Heltec-Aaron-Lee, 2020-04-09:
there already have a battery voltage reading circuit on board, did you try with this example?
[https://github.com/HelTecAutomation/ASR650x-Arduino/blob/master/libraries/Basics/examples/ADC_battery/ADC_battery.ino](https://github.com/HelTecAutomation/ASR650x-Arduino/blob/master/libraries/Basics/examples/ADC_battery/ADC_battery.ino)

nidaas, 2020-04-11:
Thanks for your response. I did and it works.
But I want to use this ADC pin for different purpose. So I wanted a way to measure internal Voltage like we can do in arduino, without any external GPIO pin.

Heltec-Aaron-Lee, 2020-04-12:
Do you mean the PVD function for STM32? As I know, ASR6501 doesn't have such a function.

Heltec-Aaron-Lee, 2020-04-12:
ASR6502 will be release soon, the ASR6502 series will have 3 ADC pins.

nidaas, 2020-04-21:
ok thanks ASR6501 doesn't have feature of monitor battery with internal component only.

closed 2020-09-18 by Heltec-Aaron-Lee