HelTecAutomation/CubeCell-Arduino #75: Missing Arduino random() overloaded functions

Kater--S, 2020-04-03:
In addition to the existing stdlib function `long random(void)` we need to implement the Arduino standard functions `long random(long max)` and `long random(long min, long max)`. 
Same for `void randomSeed(unsigned long)`.

See also [here](https://www.arduino.cc/reference/en/language/functions/random-numbers/random/).

Heltec-Aaron-Lee, 2020-04-09:
Added a `long cubecell_random(int max)` function, must rename or compile with GCC, there come with error (C langue already have a `random` function but without parameters).

Kater--S, 2020-04-09:
Shouldn't C++ allow for overloading here? Or is this in the "C" domain?
Arduino `WMath.cpp` has these definitions that rely only on an existing `random(void)` function:

```
long random(long howbig)
{
    if(howbig == 0) {
        return 0;
    }
    return random() % howbig;
}
long random(long howsmall, long howbig)
{
    if(howsmall >= howbig) {
        return howsmall;
    }
    long diff = howbig - howsmall;
    return random(diff) + howsmall;
}

```

I'm not an Arduino specialist,  but couldn't we simply incorporate `WMath.h`/`.cpp` into the package? The same approach might apply for other Arduino standard functions.

Kater--S, 2020-04-09:
Apart from this, using modulo (%) in conjunction with random number generation is not the best solution as it can affect the distribution of numbers. But that's how Arduino has implemented it ;-)