HelTecAutomation/CubeCell-Arduino #80: Offer source code of CubeCellflash

aparcar, 2020-04-21:
I can only find a binary version of the flash command, please add another repository showing the used Python script.

Heltec-Aaron-Lee, 2020-04-23:
This file contains verification and encryption algorithms, have no plan open source yet.

closed 2020-05-07 by wasn-eu

goblin, 2020-06-03:
Also CubeCellelftool is missing source code. Virustotal claims Sangfor Engine Zero detects them both as malware:

https://www.virustotal.com/gui/file/0ace4e9800e7927c53575b879d3d85a9926e299c8b3f47a9093ffd14b40108ab/details
https://www.virustotal.com/gui/file/b7dd661c8aa763e42c4f5d0cfc6f43b33ccf866df37a365a3a9f11cbe3287c08/detection

Has someone ran a more thorough scan of those binaries?

What verification/encryption is it? Does the PSoC4 even accept encrypted code? :-O

goblin, 2020-06-03:
Also, you're shipping a version of gcc, i.e. from http://119.23.153.38/download/gcc-arm-none-eabi-8-2019-q3-update-linux.tar.gz . Is it modified in any way? If so then you are **obliged** by the GPL to provide the source code of your modifications.

Heltec-Aaron-Lee, 2020-06-04:
For example, you developed a product based on CubeCell, but someone copied your hardware... If you copy the firmware from FLASH and burn it into the copied hardware, this firmware will not work.

goblin, 2020-06-04:
OK, I see, so it's for our "protection". It would be very appreciated if you could, in the future, release the source code anyway.