HelTecAutomation/CubeCell-Arduino #83: Can't install form arduino-cli because of JSON error

eni23, 2020-04-24:
Hello 

Try install the board over [arduino-cli](https://arduino.github.io/arduino-cli/getting-started/#adding-3rd-party-cores) fails with the following message:

```
Error updating index: invalid package index in http://resource.heltec.cn/download/package_CubeCell_index.json: json: invalid number literal, trying to unmarshal "\"6135813 \"" into Number
```

This is because the JSON `http://resource.heltec.cn/download/package_CubeCell_index.json: json` is not valid, there are 2 errors:
- In line `223` there is an whitespace after the number `"size": "6135813 "`
- In line `264` there is also an whitespace after the number `"size": "123811671 "`

Removing those white spaces and host the file somewhere else fixes the issue. Since the `size` fields are numbers, removing the quotation marks from all of them wold make sense.

Heltec-Aaron-Lee, 2020-04-29:
I had fixed the problem, please try again.

eni23, 2020-04-30:
unfortunately there are still whitespaces in http://resource.heltec.cn/download/package_CubeCell_index.json

Heltec-Aaron-Lee, 2020-05-04:
Hi @eni23 

Thank you, fixed again now.

closed 2020-05-04 by Heltec-Aaron-Lee

squartochi, 2022-10-03:
Hi, 
I am having the same issue.