HelTecAutomation/CubeCell-Arduino #86: Unable to load Board manager V.0.0.6 in Arduino

sivaelid, 2020-04-27:
Hi Just to report that the  Board manager does not exist in Arduino Latest version shown is  v0.0.5, even though on the Heltec Site.

Simon

Heltec-Aaron-Lee, 2020-04-29:
Hi... I can see V0.06 in my board manager:
[image]

You may need delete the cache file in your Arduino15 folder and try again, it's in this path:
[image]

Or you can use this URL: [https://github.com/HelTecAutomation/ASR650x-Arduino/releases/download/0.0.6/package_CubeCell_index.json](https://github.com/HelTecAutomation/ASR650x-Arduino/releases/download/0.0.6/package_CubeCell_index.json)

sivaelid, 2020-04-29:
Hi thanks I will try again I was using Arduino 1.8.10 but did not delete cashe file as have not needed in the past.

 

Many thanks for the great work and looking forward to the new products

 

Simon

 

 <https://user-images.githubusercontent.com/32860565/80576139-42657080-8a37-11ea-894b-18b671a23f50.png> 

 <https://user-images.githubusercontent.com/32860565/80576194-58733100-8a37-11ea-9506-357f4771c5f4.png>

sivaelid, 2020-04-29:
Hi I have just carried that out, and when loaded a new version I still only get to see  version up to V0.0.5 as below, not  version V0.0.6

 

Simon

 

 <https://user-images.githubusercontent.com/32860565/80576139-42657080-8a37-11ea-894b-18b671a23f50.png> 

 <https://user-images.githubusercontent.com/32860565/80576194-58733100-8a37-11ea-9506-357f4771c5f4.png>

Heltec-Aaron-Lee, 2020-04-29:
Did you delete the V0.0.5 package, and also deleted the cache file in Arduino15 folder?

sivaelid, 2020-04-29:
Hi yes and I tried a different version as well just now with Arduino Version 1.8.12. I will try a again with a totally clean machine

 

simon

sivaelid, 2020-04-29:
HI Arron Please not I have had to do the following.

 

*	I was using Board    http://119.23.153.38/download/package_CubeCell_index.json which  was previously on your site and  this worked to Version V0.0.5
*	I had  the Board manager in Arduino / Hardware   as well   to use the latest from GitHub.
*	After a clean install to Arduino 1.8.12 and using below It now work thanks. http://resource.heltec.cn/download/package_CubeCell_index.json

 

Simon

Heltec-Aaron-Lee, 2020-04-29:
[image]
the url [resource.heltec.cn](resource.heltec.cn) is pointed to119.23.153.38, indeed that's the same address. It should work. when I look into the json file, I can see the V0.0.6 pakcage.

[image]

sivaelid, 2020-04-29:
H Arron I just used the new board manager and giving a lot of errors related to channel mask. Here they are for info:

WARNING: library LoRa claims to run on ASR650x-Arduino architecture(s) and may be incompatible with your current board which runs on CubeCell architecture(s).

WARNING: library Sensor_ThirdParty claims to run on ASR650x-Arduino architecture(s) and may be incompatible with your current board which runs on CubeCell architecture(s).

libraries\LoRa\LoRaWan_APP.cpp.o: In function `lwan_dev_params_update':

 

C:\Users\User\AppData\Local\Arduino15\packages\CubeCell\hardware\CubeCell\0.0.6\libraries\LoRa\src/LoRaWan_APP.cpp:436: undefined reference to `userChannelsMask'

 

C:\Users\User\AppData\Local\Arduino15\packages\CubeCell\hardware\CubeCell\0.0.6/cores/asr650x/projects/CubeCellLib.a(AT_Command.o): In function `getDevParam':

 

D:\lib/AT_Command.c:1241: undefined reference to `userChannelsMask'

 

C:\Users\User\AppData\Local\Arduino15\packages\CubeCell\hardware\CubeCell\0.0.6/cores/asr650x/projects/CubeCellLib.a(AT_Command.o): In function `onUartEvent':

 

D:\lib/AT_Command.c:1082: undefined reference to `userChannelsMask'

 

C:\Users\User\AppData\Local\Arduino15\packages\CubeCell\hardware\CubeCell\0.0.6/cores/asr650x/projects/CubeCellLib.a(AT_Command.o): In function `printDevParam':

 

D:\lib/AT_Command.c:1393: undefined reference to `userChannelsMask'

 

collect2.exe: error: ld returned 1 exit status

 

exit status 1

Error compiling for board CubeCell-Board.

 

Simon

Heltec-Aaron-Lee, 2020-05-02:
These days are a national holiday. I am at home and did a complete test on the computer I used to play games (brand new computer), everything seems to be normal:
[image]

[image]

I think your problem may be because the development environment is also installed in another place. You use new examples, but the development environment and the underlying driver are still old. You need to double check :D

wasn-eu, 2020-05-07:
the problem was the missing uint16_t userChannelMask define

closed 2020-05-07 by wasn-eu