HelTecAutomation/CubeCell-Arduino #95: HTCC-AB02 pinout diagram suffers from same problem as original HTCC-AB01 pinout diagram

lnlp, 2020-05-07:
As the pinout diagram resource is external to GitHub, but relevant to both this repository and https://github.com/HelTecAutomation/HeltecDocs, I add it as an issue here.

 The [HTCC-AB02 pinout diagram](https://resource.heltec.cn/download/CubeCell/HTCC-AB02/HTCC-AB02_PinoutDiagram.pdf) suffers from the same problem as the original HTCC-AB01 pinout diagram and should be fixed _(for HTCC-AB01 it as fixed already)_.

For a description of the issue and it's importance see issue #55 for a description. The same holds for the HTCC-AB02 pinout diagram.

wasn-eu, 2020-05-07:
fixed

lnlp, 2020-05-07:
@wasn-eu 
> fixed

No, this has not been fixed yet.

wasn-eu, 2020-05-07:
There are now the mentilned red lines showing the position of the pins.
If that is not what you thought of please explain

lnlp, 2020-05-07:
Please see #55 discussions for a description.

I still see this pinout diagram for HTCC-AB02:

[image]

While I expect something similar to this: Check and compare the **positions** of the GPIO/pin number**labels** and red lines for the User Control and VExt pins:

[image]

Last image is from current HTCC-AB01 pinout (which was previously corrected),
which uses different GPIO and pin numbers. This image is just used for an example.

wasn-eu, 2020-05-07:
If i open

https://resource.heltec.cn/download/CubeCell/HTCC-AB02/HTCC-AB02_PinoutDiagram.pdf

There are red lines for those pins

lnlp, 2020-05-07:
Please _carefully_ read my above responses and all responses in issue #55 _again_.

It _**very clearly**_ describes the issue and the pictures very clerly describe the issue.

Apparently many details of above responses get lost in English to Chinese translation?
The pictures are not translated and also clearly show the issue.

wasn-eu, 2020-05-07:
please stay calm, no need to get rude.

for people used to read those sheets everyday this is clear, but i fully understand that there is a potential for missunderstanding.

you only want to separate the labels for those pins with red lines from the other labels?

I will reopen the issue but we have to wait for somebody from heltec to change it because i dont have the rights to change it on the website.

lnlp, 2020-05-07:
I _am_ calm and I am not rude.

Thanks for reopening.

Yes there is a big potential for missunderstanding, included for people used to reading such pinouts. 

This was already clearly addressed in issue #55 but then for HTCC-AB01.
The same problem is now repeated in pinout for HTCC-AB02, which I don't understand because one should have learned from #55 how not to do it.

wasn-eu, 2020-05-08:
do you think this version is better:
[image]

lnlp, 2020-05-09:
Yes, much better.

wasn-eu, 2020-05-09:
That was only a quick photoshop fix.
@Heltec-Aaron-Lee please make this changes in the PDF