HelTecAutomation/CubeCell-Arduino #103: [BUG] No LoRa examples

IoTopenTech, 2020-05-29:
Hi, I have installed the board manager in Arduino 1.8.10, but there is no LoRa category in examples
[image]

Heltec-Aaron-Lee, 2020-05-31:
Seems every thing is normal at my side:
[image]

Do you have another library named "LoRa"?

IoTopenTech, 2020-05-31:
Yes, that is the problem... I had another librart named LoRA.
Thanks.

closed 2020-05-31 by IoTopenTech

JuriShalkin, 2020-10-28:
> Yes, that is the problem... I had another librart named LoRA.
> Thanks.

How did you solve this problem? I also have no examples of onewire.