HelTecAutomation/CubeCell-Arduino #105: Please provide source code of the GCC binaries you're shipping

goblin, 2020-06-03:
You're shipping a version of gcc, i.e. from http://119.23.153.38/download/gcc-arm-none-eabi-8-2019-q3-update-linux.tar.gz . Is it modified in any way? Where is the source code? How can I rebuild it?

_Originally posted by @goblin in https://github.com/HelTecAutomation/ASR650x-Arduino/issues/80#issuecomment-638211474_

Heltec-Aaron-Lee, 2020-06-04:
Haha, you are overestimating me 🤣 

I think I don’t have the ability to modify GCC, but because of the Great Firewall in mainland China, many users in Mainland China cannot download or the speed is very slow, so I store GCC on my own server.

goblin, 2020-06-04:
OK, thanks for the info :-)

I had the same problem downloading from Europe. The `get.py` and the Arduino IDE were trying to download the GCC from your server, but the connection was stalling after getting a few MB; presumably this was due to the Great Firewall kicking in. I had to bypass these scripts and download manually with `wget --read-timeout=15`.

But if I can use the `gcc` shipped by Debian, then it's fine.

closed 2020-06-04 by goblin