HelTecAutomation/CubeCell-Arduino #116: Sketch is not compiles and CubeCell.ino.hex not found

Buzda, 2020-07-19:
Hi there!

I tried your library on my mac and it works fine. When I tried it on my windows 10 laptop, it throws the error below. I found some workarounds for this by disabling the McAfee antivirus as it seems to be the reason. I tried doing this but I am still getting the same error. Has anyone been able to solve this issue?
Thanks!

Arduino: 1.8.13 (Windows Store 1.8.39.0) (Windows 10), Board: "CubeCell-Board（HTCC-AB01）, REGION_AS923, CLASS_A, OTAA, ON, CONFIRMED, OFF, ON, ACTIVE, None"

THE ERROR:

'C:\Users\X' is not recognized as an internal or external command,

operable program or batch file.

[9576] Failed to execute script CubeCellelftool

Traceback (most recent call last):

  File "CubeCellelftool.py", line 3, in <module>

  File "elftool.py", line 26, in elftool.main

FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Users\\X~1\\AppData\\Local\\Temp\\arduino_build_151539/CubeCell.ino.hex'

exit status -1

Error compiling for board CubeCell-Board（HTCC-AB01）.

Invalid version 'Internal RTC2.0' for library in: C:\Users\X X X\Documents\Arduino\libraries\ESP32_LoRaWAN-master

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Heltec-Aaron-Lee, 2020-08-13:
It seems the "~" in username "X~1" can't be recognized

Buzda, 2020-08-13:
I abandoned arduino and worked on platformIO on VScode and it is working fine now. It is just the combination of CubeCell and Arduino that was the problem.

wasn-eu, 2020-09-06:
Do you have a space in your username?
Spaces and special characters are making problems

alexcorvis84, 2020-09-30:
> Do you have a space in your username?
> Spaces and special characters are making problems

I can confirm this Issue is happening. Today I was trying to compile some of the examples without luck and getting the dame error.

4 hours after trying different configs (i thought It was related to libraries at first glance...) I checked this Issue. Then I tried to config Arduino IDE and framework with a user without spaces and It worked like a charm.

So, can be this 'error' fixed somehow? Probably more users have username with spaces and its frustrating lost a lot of time debugging until you find out whats really happening.

Would be appreciated if It can be fixed 😉

ruddyBIT, 2020-11-11:
buenas tardes **alexcorvis84** tengo el mismo problema en Windows 10, adquirí  dos placas HELTECTEC CubCell HTCC-AB01 para realizar comunicación punto a punto. Cambie el nombre de usuario de mi Lap top y sigo con el mismo problema, puedes compartir mas detalles para corregir el error. 
les adjunto la foto del error que tengo, ya llevo varios días con este problema, agradezco anticipadamente cualquier ayuda. 
[image]
Arduino: 1.8.13 (Windows 10), Tarjeta: "CubeCell-Board （HTCC-AB01）, REGION_US915, CLASS_A, CUSTOM, OTAA, ON, CONFIRMED, OFF, ON, ACTIVE, None"

“C: \ Users \ mi” no se reconoce como un comando interno o externo,
programa o archivo por lotes ejecutables.
[10540] Error al ejecutar el script CubeCellelftool
Traceback (última llamada más reciente):
Archivo “CubeCellelftool.py”, línea 3, en
Archivo “elftool.py”, línea 26, en elftool.main
FileNotFoundError: [Errno 2] No existe tal archivo o directorio: 'C: \ Users \ MIPC ~ 1 \ AppData \ Local \ Temp \ arduino_build_785592 / LoRaSender.ino.hex'
estado de salida -1
Error compilando para la tarjeta CubeCell-Board （HTCC-AB01）.