HelTecAutomation/CubeCell-Arduino #117: printf is broken

eiten, 2020-07-29:
Hello there. Printf is broken/not working for all format specifications. Sketch:
```c++
void setup() {
  Serial.begin(115200);
  Serial.printf("Unsigned char: %hhu, unsigned long: %hu", (uint8_t)123, (uint16_t)1234);
}

void loop() {
}
```
expected output (confirmed on your LoRa 32 ESP32 board):
`Unsigned char: 123, unsigned long: 1234`
Output if compiling for and running on CubeCell board:

`Unsigned char: hu, unsigned long: 123`
Same wrong behaviour on Arduino IDE and platform.io.

Regards, Edi

Heltec-Aaron-Lee, 2020-08-13:
sorry, %hhu is no supported, the printted 123 is for (uint8_t)123,  and (uint16_t)1234 is ignored

DavidStacer, 2021-06-07:
Is the %f format for float or double supported?

 float test_float = 45.6;
 Serial.printf("test_float is %f \n", test_float);  results in some hex character being printed out.

  Serial.print("test_float ");
  Serial.println(test_float);     results in the proper output of 45.6 on the monitor display.

What format specifiers are supported for a printf?