HelTecAutomation/CubeCell-Arduino #118: Serial.flush() is broken

eiten, 2020-07-29:
Serial.flush is skipping the last 8 characters of serial output! Sketch:
```c++
void setup() {
  Serial.begin(115200);
  Serial.println("Sending some text that needs to be displayed, It's a rather long text, so it is still sending whern Serial.flush() kicks in");
  Serial.flush();
  Serial.println();
  Serial.println("Also with shorter texts");
  Serial.flush();
} 

void loop() {
}
```
Correct output on a LoRa 32:
```
Sending some text that needs to be displayed, It's a rather long text, so it is still sending whern Serial.flush() kicks in

Also with shorter texts
```
Broken output on a CobeCell board:
```
Sending some text that needs to be displayed, It's a rather long text, so it is still sending whern Serial.flush() k
Also with shorte
```

lyusupov, 2020-09-03:
I would also vote for the issue to be resolved.

[Current implementation of HardwareSerial::flush()](https://github.com/HelTecAutomation/ASR650x-Arduino/blob/master/cores/asr650x/Serial/HardwareSerial.cpp#L152) is doing this:

```
void HardwareSerial::flush()
{
	if( _uart_num == UART_NUM_0)
	{
		UART_1_SpiUartClearRxBuffer();
		UART_1_SpiUartClearTxBuffer();
	}
	else
	{
		UART_2_SpiUartClearRxBuffer();
		UART_2_SpiUartClearTxBuffer();	
	}
}
```

it is Ok to clean Rx buffer.
However, according to Arduino [Serial.flush() definition](https://www.arduino.cc/reference/en/language/functions/communication/serial/flush/) this function has to wait for the transmission of outgoing serial data to complete.

The issue with current ASR650x flush() implementation is that it forces to clear Tx buffer instead of waiting for the buffer to become empty by regular data flow.

lyusupov, 2020-09-03:
I've created a PR with a proposed fix candidate: https://github.com/lyusupov/ASR650x-Arduino/commit/95afd5d65ee8f9572a3a9edc931c95aca737a0d4
It works nicely for me.

However it makes sense to test the FC in order to make sure that it will never do infinite loop inside the flush().

closed 2020-10-09 by Heltec-Aaron-Lee

eiten, 2023-06-02:
Well, it's still broken, check the following sketch (it removes the last char):
```
#include <Arduino.h>

void setup()
{
  Serial.begin(115200);
}

void loop()
{
  Serial.print("0123456789");
  Serial.flush();
  Serial.println();
  delay(1000);
}
```
Output is
```
012345678
012345678
012345678
```
For me, I fixed it by removing the `UART_1_SpiUartClearTxBuffer();` and `UART_2_SpiUartClearTxBuffer();` lines (#320 and #326) from
[HardwareSerial.cpp](https://github.com/HelTecAutomation/CubeCell-Arduino/blob/1b5829bcc499cd2a16c023ce03a9f5dd9ecfe450/cores/asr650x/Serial/HardwareSerial.cpp#LL312C1-L328C2).
These lines seem to be redundant, as in the line before, you check if the buffer is empty.

Quency-D, 2023-06-05:
@eiten 
Your suggestion is correct. We have updated the relevant functions. Thank you.

eiten, 2023-06-13:
Thanks a lot! I'll give feedback as soon as I will be able to test it.