HelTecAutomation/CubeCell-Arduino #119: ARDUINO define missing

eiten, 2020-07-29:
Many libs depend on the ARDUINO define to figure out what the api is or which headers to include. Example:
```C++
#if ARDUINO >= 100
#include "Arduino.h"
#include "Print.h"
#else
#include "WProgram.h"
#endif
```
This define is missing in ASR650x-"Arduino". Please fix, I assume it should be 10805.

gazzyt, 2020-07-30:
Just ran into this issue myself today.

gazzyt, 2020-07-30:
I had v0.0.3 of this installed for some reason. Updating to v1.0.0 fixed this for me

eiten, 2020-07-31:
Well, i have 1.0.0 installed on PlatformIO where this does not work

eiten, 2020-08-03:
I gotta correct myself. It's working in Arduino IDE, not working in Platform.io

tuxphone, 2020-09-15:
You can circumvent the problem with adding the following line to platform.io:

`build_flags = -D ARDUINO=100`

Heltec-Aaron-Lee, 2020-09-18:
Now have V1.1.1 release, did you try that?
[image]

closed 2020-10-09 by Heltec-Aaron-Lee